cleanflight/cleanflight #388: Want to help, where to find the community?

avoid3d, 2015-01-20:
Hey, I want to start contributing to this project, but where is the discussion? Is there a wiki or a forum where I can start?

Sorry to use issues for this, but I don't know where to 'join' :P

thenickdude, 2015-01-20:
There's a bunch of user discussion on RCGroups in this thread:

http://www.rcgroups.com/forums/showthread.php?t=2249574

Most of the developers are hanging out on IRC, details about that channel are in the readme:

https://github.com/cleanflight/cleanflight/blob/master/README.md

fiveangle, 2015-01-21:
This used to be a spot, but has seen less activity as of late:  http://www.multiwii.com/forum/viewtopic.php?f=23&t=5149

avoid3d, 2015-01-21:
Thanks guys, been on IRC a lot and that's helped :)

closed 2015-01-21 by avoid3d