cleanflight/cleanflight #389: No SiLabs VCP driver for Mac OS X - Yosemite

avoid3d, 2015-01-20:
I bought a naze 32 from hobbyking, and the board comes with a SiLabs usb to uart bridge, which needs drivers for mac.

I have not been able to get these working with Yosemite.

Here is a request: http://community.silabs.com/t5/Feature-Requests/Mac-OS-X-10-10-Yosemite-VCP-driver-any-time-soon/td-p/133815

Here is a driver RC: https://community.silabs.com/t5/Interface/OS-X-10-10-CP210x-VCP-Driver-Release-Candidate/m-p/138472#U138472

avoid3d, 2015-01-20:
So it turns out other users, @thenickdude for example have had success with the old driver from the drivers page.

I may have been using a 'charge only' usb cable, will update tonight.

thenickdude, 2015-01-20:
Yeah I'm running com.silabs.driver.CP210xVCPDriver64 (3.1.0d1) and having no issues on Yosemite.

fiveangle, 2015-01-21:
Same here: 3.10d1 working fine on Yosemite.

avoid3d, 2015-01-21:
I resolved this by using a different usb cable.

closed 2015-01-21 by avoid3d