cleanflight/cleanflight #390: Add debug variable to blackbox recording

tracernz, 2015-01-21:
@thenickdude It would be really helpful if the debug variable ([defined line 88 of mw.c](https://github.com/cleanflight/cleanflight/blob/5be2b9229d418271bbe3cb896f025e2f90b126b4/src/main/mw.c#L88)) were logged by blackbox and decoded by blackbox_decode, don't you think? Also, perhaps current would be nice too.

thenickdude, 2015-01-21:
The thing about logging variables is that if you sample them at the wrong
time, you don't get any useful information. There's no good time to sample
a generic "debug" array (that could contain anything) that would suit all
uses of it. Typically what you really want is to log an event, which
happens at a particular time and might even happen multiple times per loop
iteration, or not at all on some iterations. Multiple distinct events can
happen per logging cycle. The event logging system, which has provision for
adding custom events with custom payloads, would do a much better job than
the debug array.

I definitely want to add current, just have to find someone to test it out
for me. Do you have a current sensor?

thenickdude, 2015-01-21:
On the other hand, I could see the value in adding a generic sort of logged
debug event which would allow it to be decoded with no changes to
blackbox_decode (i.e. log an array of values of specified length much like
the debug array), so I'll add an event type for that purpose!

tracernz, 2015-01-21:
Okay, I'll look into the event system. A generic debug event which can log a small array of values would be great. The purpose that I have in mind currently is to compare the current measured by ADC from the current sensor with my new virtual current sensor feature. I'm sure there are many other things we can do with it during future development.

My openlogs should arrive in the next couple of days and yes I have a current sensor on my main quad. I'd be happy to test current logging capability if you add it. :)

ledvinap, 2015-01-21:
I'll work on current sensor soon ...

using debug is not that great idea  - it is reused, so only torso of debug capability remain in code. It would be much cleaner to have some easy way to annotate variables for  'debugging'/logging and allow other developers to enable logging easily.

With current blackbox implementation it should be quite easy - just use #define to enable variable logging (if logging on loop time is appropriate) .

Similar approach may be used for 'debug[]' - '#define DEBUG_SBUS_ERRORS DEBUG_CH1' in one place may work fine (for single-bit event I use https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/pin_debug.h)

hydra, 2015-05-15:
I agree with @ledvinap and @thenickdude here, logging the debug variables isn't really what is needed due to the rate of sampling.

@thenickdude Can you help out?  No hurry.

digitalentity, 2015-05-26:
@thenickdude I +1 for this issue. Currently I ran into issues with new GPS PH and don't have any means to get the insides of my code while flying. On the bench everything look normal, but in flight - oops, it doesn't work at all...
Having a way to log an arbitrary value to blackbox would greatly help!

ledvinap, 2015-05-26:
@digitalentity : It's quite easy to add new logged variable. Just grep existing one and add necessary code + definitions in all places.
Decoding tools will handle new variable without modification (at least blackbox_decode)

digitalentity, 2015-05-27:
@ledvinap thanks, will try it

digitalentity, 2015-05-27:
@ledvinap Thanks! Adding new vars to blackbox is indeed very easy!

closed 2017-02-25 by hydra