cleanflight/cleanflight #391: Feature Helicopter Mix

mmartinmate, 2015-01-21:
Hello,

First of all congratulations for the very good job you have done.

I would ask if you are planned to include mixes for traditional helicopters with and without flybar?

I have seen that APM includes this feature.

And TauLabs/OpenPilot includes also the feature but 'only' for flybarless helicopters.

I don't know if that is very complex or could be done easily.

Yours sincerely,
        Miguel.

dogz85, 2015-02-15:
+1 I would love to be able to use the afro mini as a flybarless gyro and stabilisation control.  never flown a heli but if this was implemented i would do it  :)

tracernz, 2015-02-15:
I might look into this if I have time. I've got an unloved HK450 lying around somewhere I could try it out on...

matvey83, 2015-02-26:
This would be a great feature.

bbrummer, 2015-03-15:
I'd *love* this as well.  I've been trying to get TauLabs to run on an AfroMini specifically to use it as a FBL controller to replace the "3-in-1" board on my Blade Nano CPX.  The boards on these little helis fail easily and often and are crazy expensive to replace.  The issue is size and weight...every .1 of a gram matters.

On a lark I tried using CleanFlight and making it pretend it's a tri-copter in 3D mode as that logically should respond much like a 120 degree swash plate (although I need a 140 degree) and even the tail servo should mostly respond the same.  But so far...not so great.  I think it needs a real helicopter mix (with PWM motor tail!) to be usable, but it doesn't seem like it should take that much to make it happen...  Too bad cmix doesn't handle servos :(

hydra, 2015-05-15:
I would welcome a PR to add this.

mmartinmate, 2015-05-15:
Hi Dominic,

I do not undertand what means PR.
I have two naze32 clone one with magnetometer and barometer and also a
CC3D. A Crius ublox 6M GPS and a Sirf III that works if I modify code to
config it with NMEA and 115200 bauds and also bluetooth and 3DR Radio
modules.
But I does not have flybarless helicopter.

If I could help tell me what I can do.

Best regards,

      Miguel.

tracernz, 2015-05-15:
PR means Pull Request. It's the GitHub way of merging changes from one branch to another. When adding new features the code is usually developed in a separate branch (typically in a separate fork belonging to the developer in question) and then merged to the main branch through a pull request.

Sorry to say I no longer have any intention of working on this as I've moved on to other projects.

jaydlawrence, 2016-08-19:
It has been over a year since this started. Has anyone made any progress on this?

I am looking at getting my first heli and coming from multirotors, it seems like such a rippoff that the "flybarless control systems" cost so much money when we have far more able flight controllers for an order of magnitude cheaper for quads.

jcg1541, 2017-01-07:
My bet is that heli boards will never make it into mainstream software projects like cleanflight, taulab, due to the economical deviation of the heli design. Someone did some calculations and said that the economical gain is in favour of quad design when the vehicle weight is under certain threshold. Something to do with weight gain in the 3rd order while size increases in the 1st order. That is why flying cars with multi rotors will be suppressed forever, no matter how you hype it. One sentence, laws of economy.

jcg1541, 2017-01-09:
We might need to look into CC3D chip/board. It already has a production heli Esky 150X . I just ordered one on banggood.

closed 2017-02-25 by hydra