cleanflight/cleanflight #392: Impossible to connect cleanflight to my Naze with display connected

Malaquitte, 2015-01-21:
Hi,

I've just received my oled display this afternoon. It's the same as this video https://www.youtube.com/watch?v=Z1MrwPKicGc. Unfortunatly when it's plugged I can't connect cleanflight to Naze board. Here is the log:

18:41:02 -- Unique device ID received - 
18:41:47 -- Serial port successfully closed
18:41:49 -- Serial port successfully opened with ID: 9
18:41:59 -- No configuration received within 10 seconds, communication failed

If I enable DISPLAY Feature in CLI without Display connected and then reboot Naze board with display plugged, it doesn't works either.

Here is my features, all is working: 

# features
Enabled features: RX_PPM VBAT MOTOR_STOP GPS FAILSAFE SONAR TELEMETRY CURRENT_METER RSSI_ADC DISPLAY

Thanks for your help.

hydra, 2015-01-21:
what happens if you reset the configuration to default using `defaults` cli command command and then just enabling the display feature?

Malaquitte, 2015-01-22:
I've reset the configuration and now I can connect cleanflight. I've just activated Display command but my display is still black. 
I've also check again my Naze/Display,VCC & GND connections.
Did I missed something else before thinking that I received a broken display ?

hydra, 2015-01-22:
if you have an oscilioscope or logic analyzer you can check the SDA/SCL signals going to the display to see if the display 'ack'ing the data from the FC.  You could also try hooking the display up to an ardiuno and running a test program too.

Not really much I can do about your problem since i have no way to reproduce the issue here - i have 3 different types of screens and they all work just fine so the problem is likely to be with your display or FC.

Please remember the issue tracker is not for support.  Use IRC or the forums first, then if the problem is confirmed create issues detailing how to repeat the problem.

closed 2015-01-22 by hydra