cleanflight/cleanflight #393: Trouble building cleanflight.

avoid3d, 2015-01-21:
If I cd to cleanflight and run:
'make clean'
'make TARGET=NAZE32'

I get a lot of output showing files compiling, and then linking breaks. Here is the output:

avoidd-mbp:cleanflight avoidd$ make TARGET=NAZE
%% startup_stm32f10x_md_gcc.S
%% accgyro_adxl345.c
%% accgyro_bma280.c
%% accgyro_l3g4200d.c
%% accgyro_mma845x.c
%% accgyro_mpu3050.c
%% accgyro_mpu6050.c
%% accgyro_spi_mpu6500.c
%% adc.c
%% adc_stm32f10x.c
%% barometer_bmp085.c
%% barometer_ms5611.c
%% bus_spi.c
%% bus_i2c_stm32f10x.c
%% compass_hmc5883l.c
%% display_ug2864hsweg01.c
%% gpio_stm32f10x.c
%% inverter.c
%% light_led_stm32f10x.c
%% light_ws2811strip.c
%% light_ws2811strip_stm32f10x.c
%% sonar_hcsr04.c
%% pwm_mapping.c
%% pwm_output.c
%% pwm_rx.c
%% serial_softserial.c
%% serial_uart.c
%% serial_uart_stm32f10x.c
%% sound_beeper_stm32f10x.c
%% system_stm32f10x.c
%% timer.c
%% timer_stm32f10x.c
%% hardware_revision.c
%% autotune.c
%% navigation.c
%% gps_conversion.c
%% colorconversion.c
%% gps.c
%% ledstrip.c
%% display.c
%% telemetry.c
%% frsky.c
%% hott.c
%% msp.c
%% smartport.c
%% sonar.c
%% barometer.c
%% blackbox.c
%% build_config.c
%% version.c
%% config.c
%% runtime_config.c
%% maths.c
%% printf.c
%% typeconversion.c
%% main.c
%% mw.c
%% altitudehold.c
%% failsafe.c
%% flight.c
%% imu.c
%% mixer.c
%% bus_i2c_soft.c
[... 42 lines cut ...]
%% stm32f10x_spi.c
%% stm32f10x_tim.c
%% stm32f10x_usart.c
%% stm32f10x_wwdg.c
arm-none-eabi-gcc -o obj/main/cleanflight_NAZE.elf obj/main/NAZE/startup_stm32f10x_md_gcc.o obj/main/NAZE/drivers/accgyro_adxl345.o obj/main/NAZE/drivers/accgyro_bma280.o obj/main/NAZE/drivers/accgyro_l3g4200d.o obj/main/NAZE/drivers/accgyro_mma845x.o obj/main/NAZE/drivers/accgyro_mpu3050.o obj/main/NAZE/drivers/accgyro_mpu6050.o obj/main/NAZE/drivers/accgyro_spi_mpu6500.o obj/main/NAZE/drivers/adc.o obj/main/NAZE/drivers/adc_stm32f10x.o obj/main/NAZE/drivers/barometer_bmp085.o obj/main/NAZE/drivers/barometer_ms5611.o obj/main/NAZE/drivers/bus_spi.o obj/main/NAZE/drivers/bus_i2c_stm32f10x.o obj/main/NAZE/drivers/compass_hmc5883l.o obj/main/NAZE/drivers/display_ug2864hsweg01.o obj/main/NAZE/drivers/gpio_stm32f10x.o obj/main/NAZE/drivers/inverter.o obj/main/NAZE/drivers/light_led_stm32f10x.o obj/main/NAZE/drivers/light_ws2811strip.o obj/main/NAZE/drivers/light_ws2811strip_stm32f10x.o obj/main/NAZE/drivers/sonar_hcsr04.o obj/main/NAZE/drivers/pwm_mapping.o obj/main/NAZE/drivers/pwm_output.o obj/main/NAZE/drivers/pwm_rx.o obj/main/NAZE/drivers/serial_softserial.o obj/main/NAZE/drivers/serial_uart.o obj/main/NAZE/drivers/serial_uart_stm32f10x.o obj/main/NAZE/drivers/sound_beeper_stm32f10x.o obj/main/NAZE/drivers/system_stm32f10x.o obj/main/NAZE/drivers/timer.o obj/main/NAZE/drivers/timer_stm32f10x.o obj/main/NAZE/hardware_revision.o obj/main/NAZE/flight/autotune.o obj/main/NAZE/flight/navigation.o obj/main/NAZE/flight/gps_conversion.o obj/main/NAZE/common/colorconversion.o obj/main/NAZE/io/gps.o obj/main/NAZE/io/ledstrip.o obj/main/NAZE/io/display.o obj/main/NAZE/telemetry/telemetry.o obj/main/NAZE/telemetry/frsky.o obj/main/NAZE/telemetry/hott.o obj/main/NAZE/telemetry/msp.o obj/main/NAZE/telemetry/smartport.o obj/main/NAZE/sensors/sonar.o obj/main/NAZE/sensors/barometer.o obj/main/NAZE/blackbox/blackbox.o obj/main/NAZE/build_config.o obj/main/NAZE/version.o obj/main/NAZE/config/config.o obj/main/NAZE/config/runtime_config.o obj/main/NAZE/common/maths.o obj/main/NAZE/common/printf.o obj/main/NAZE/common/typeconversion.o obj/main/NAZE/main.o obj/main/NAZE/mw.o obj/main/NAZE/flight/altitudehold.o obj/main/NAZE/flight/failsafe.o obj/main/NAZE/flight/flight.o obj/main/NAZE/flight/imu.o obj/main/NAZE/flight/mixer.o obj/main/NAZE/drivers/bus_i2c_soft.o obj/main/NAZE/drivers/serial.o obj/main/NAZE/drivers/sound_beeper.o obj/main/NAZE/drivers/system.o obj/main/NAZE/io/beeper.o obj/main/NAZE/io/rc_controls.o obj/main/NAZE/io/rc_curves.o obj/main/NAZE/io/serial.o obj/main/NAZE/io/serial_cli.o obj/main/NAZE/io/serial_msp.o obj/main/NAZE/io/statusindicator.o obj/main/NAZE/rx/rx.o obj/main/NAZE/rx/pwm.o obj/main/NAZE/rx/msp.o obj/main/NAZE/rx/sbus.o obj/main/NAZE/rx/sumd.o obj/main/NAZE/rx/sumh.o obj/main/NAZE/rx/spektrum.o obj/main/NAZE/rx/xbus.o obj/main/NAZE/sensors/acceleration.o obj/main/NAZE/sensors/battery.o obj/main/NAZE/sensors/boardalignment.o obj/main/NAZE/sensors/compass.o obj/main/NAZE/sensors/gyro.o obj/main/NAZE/sensors/initialisation.o obj/main/NAZE/core_cm3.o obj/main/NAZE/system_stm32f10x.o obj/main/NAZE/misc.o obj/main/NAZE/stm32f10x_adc.o obj/main/NAZE/stm32f10x_bkp.o obj/main/NAZE/stm32f10x_dac.o obj/main/NAZE/stm32f10x_dbgmcu.o obj/main/NAZE/stm32f10x_dma.o obj/main/NAZE/stm32f10x_exti.o obj/main/NAZE/stm32f10x_flash.o obj/main/NAZE/stm32f10x_fsmc.o obj/main/NAZE/stm32f10x_gpio.o obj/main/NAZE/stm32f10x_i2c.o obj/main/NAZE/stm32f10x_iwdg.o obj/main/NAZE/stm32f10x_pwr.o obj/main/NAZE/stm32f10x_rcc.o obj/main/NAZE/stm32f10x_rtc.o obj/main/NAZE/stm32f10x_sdio.o obj/main/NAZE/stm32f10x_spi.o obj/main/NAZE/stm32f10x_tim.o obj/main/NAZE/stm32f10x_usart.o obj/main/NAZE/stm32f10x_wwdg.o -lm -nostartfiles --specs=nano.specs -lc -lnosys -mthumb -mcpu=cortex-m3 -flto -fuse-linker-plugin -Os -ggdb3 -static -Wl,-gc-sections,-Map,./obj/main/cleanflight_NAZE.map -T./src/main/target/stm32_flash_f103_128k.ld
`HardFault_Handler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/main.o (symbol from plugin)
`SysTick_Handler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/system.o (symbol from plugin)
`EXTI1_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/sonar_hcsr04.o (symbol from plugin)
`DMA1_Channel4_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/serial_uart_stm32f10x.o (symbol from plugin)
`DMA1_Channel6_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/light_ws2811strip_stm32f10x.o (symbol from plugin)
`EXTI9_5_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/sonar_hcsr04.o (symbol from plugin)
`TIM1_UP_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/timer.o (symbol from plugin)
`TIM1_CC_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/timer.o (symbol from plugin)
`TIM2_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/timer.o (symbol from plugin)
`TIM3_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/timer.o (symbol from plugin)
`TIM4_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/timer.o (symbol from plugin)
`I2C1_EV_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/bus_i2c_stm32f10x.o (symbol from plugin)
`I2C1_ER_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/bus_i2c_stm32f10x.o (symbol from plugin)
`I2C2_EV_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/bus_i2c_stm32f10x.o (symbol from plugin)
`I2C2_ER_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/bus_i2c_stm32f10x.o (symbol from plugin)
`USART1_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/serial_uart_stm32f10x.o (symbol from plugin)
`USART2_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/serial_uart_stm32f10x.o (symbol from plugin)
`EXTI15_10_IRQHandler' referenced in section `.isr_vector' of obj/main/NAZE/startup_stm32f10x_md_gcc.o: defined in discarded section `.text' of obj/main/NAZE/drivers/barometer_bmp085.o (symbol from plugin)
collect2: error: ld returned 1 exit status
make: *** [obj/main/cleanflight_NAZE.elf] Error 1

avoid3d, 2015-01-21:
Fixed, was using the wrong compiler binary.

closed 2015-01-21 by avoid3d

fiveangle, 2015-01-21:
Generally, it's best to clarify what fixes an issue (e.g. Using version xxx of yyy when version zzz is required). "Fixed" doesn't help others :)

Otherwise, just ask these questions on IRC or the forums (can you tell I'm about to embark on building this ? ;)

-=dave