cleanflight/cleanflight #395: Decouple math utility code from the flight code

avoid3d, 2015-01-21:
There are functions like normalizeV and DEGREES_TO_DECIDEGREES in main/flight which should move into main/common/maths.

First I am going to write tests, then try move them, is there some reason not to go ahead with this plan?

hydra, 2015-01-22:
yes, it's fine to move math code into a the common maths code.  tests would be great too.

avoid3d, 2015-01-22:
Awesome, I have some tests for the math code, will push later :)

hydra, 2015-03-03:
@avoid3d are you working more on this or shall we close it?

avoid3d, 2015-03-03:
Hey, thanks for the ping, been very busy moving country, keen to get started coding and flying again soon, will definitely be working more on this!

hydra, 2015-05-15:
@avoid3d any update, are you working on this or can anyone work on it?

closed 2017-02-25 by hydra