cleanflight/cleanflight #407: External mag alignment improvement

strips, 2015-01-22:
I hava a Flip32 (acro) and an external hmc5883. The mag is mounted under the GPS PCB and because of the connector pins I have rotated it 90deg CW so I think I should use CW90_DEG_FLIP.

It looks like this improvment has been asked before but not really fixed: https://github.com/cleanflight/cleanflight/issues/101

It looks to me when a hmc5883 and Naze is discovered CW180_DEG is hard coded:
```c
#ifdef MAG
    if (hmc5883lDetect()) {
#ifdef NAZE
        magAlign = CW180_DEG;
#endif
    } else {
        sensorsClear(SENSOR_MAG);
    }
#endif
```

I'm thinking to look into this my self as it should be a simple fix to test. But need to set up a development environment first.

hydra, 2015-01-22:
You can change the alignment on a per sensor basis already, see align_mag command in the cli.

Closing this ticket, but perhaps you can create a pull request to document how the `align_*` settings work?

closed 2015-01-22 by hydra

strips, 2015-01-22:
Wow, sorry, my bad! I did a few searches and never found the docs. The docs already cover this. https://github.com/multiwii/baseflight/wiki/CLI-Variables

I saw my self blind on the code snippet I found referenced in the other issue. Didn't realize it was old code.