cleanflight/cleanflight #409: Switching profiles inflight

ledvinap, 2015-01-22:
Is there any reason why not to switch profiles in-flight?

When testing, it could work as failsafe - configure one profile as safe and prepare changes to test in another profile(possibly activating them safe distance from ground). If something goes wrong(failed autotune ... ), return to safe and land ..

hydra, 2015-01-23:
it could be disaterous if mode ranges and other settings for arming are not synced between profiles.

ledvinap, 2015-01-23:
Yes, that can be fun. But it can be tested on ground, so it may be reasonable to take this risk...

hydra, 2015-05-15:
I don't think we should do this, the method to change profiles would be via in-flight adjustments, but if the adjustments were not configured in the second profile it would likely be bad.

I'll close this but will enter into further discussion as required and would review a PR but i think there are better things to spend time on at the moment.

closed 2015-05-15 by hydra

ledvinap, 2015-05-15:
OK . any other idea how to implement 'save-me' mode?
When testing new code, it's quite easy to do something stupid ... It would be great to have mechanism how to get quad into flyable state to land it safely ...

dascons, 2015-05-17:
When bad things happen then force angle mode?

ledvinap, 2015-05-17:
@dascons : PID settings (or whole PID) may be unflyable, so angle mode may not help (and even make things worse).

dascons, 2015-05-17:
Then the only real thing I can think of is just cut throttle and hope,
sorry for being so shallow in the thinking