cleanflight/cleanflight #412: CLI command "R" will not activate bootloader on STM32F3 based targets

MJ666, 2015-01-23:
Looks to be the related code is not properly working? If executed the target will make an normal reboot without activating the bootloader. This will be the first step to be able to flash the device in DFU mode via the GUI without using the boot pins.
 
Here is the related function call in "serial_uart_stm32f30x.c":

34 void systemResetToBootloader(void) { 
35     // 1FFFF000 -> 20000200 -> SP 
36     // 1FFFF004 -> 1FFFF021 -> PC 
37 
38     *((uint32_t *)0x20009FFC) = 0xDEADBEEF; // 40KB SRAM STM32F30X 
39 
40     systemReset(); 
41 } 

Likely the address for this magic is not correct? Unfortunately documentation for this feature looks to be very limited. Still investigating and do some testing.

tracernz, 2015-01-23:
Since I'm curious I did a little Googling. I came up with this video (https://www.youtube.com/watch?v=cvKC-4tCRgw) which seems to require quite a bit more code than what's included in your snippet above. See page 125 of http://www.st.com/web/en/resource/technical/document/application_note/CD00167594.pdf for the address.
Hope this helps!

MJ666, 2015-01-23:
I have seen this already and this is may be a alternative method and we can go for this if we can not get the other method to work.

In my understanding the systemReset() will make a complete MCU reset and takes care on resetting all peripherals. The magic code (0xDEADBEEF) will survive the reset in ram and will trigger the bootloader activation. This is working quite well on F1 based targets. My guess is that the magic code is place at a wrong address.

MJ666, 2015-01-25:
Related fix is merged. Closed.

closed 2015-01-25 by MJ666