cleanflight/cleanflight #420: tabs disappear when connect?

Vincent81, 2015-01-24:
Hello
no connection :
[image]
With connection  :
[image]

tracernz, 2015-01-24:
Yes, I'm also getting this on OS X with the newest configurator.

fiveangle, 2015-01-24:
Works for me.  Is it intermittent or do they never show up ?

-=dave

thenickdude, 2015-01-24:
Works for me on OS X on 0.60.0 and 0.60.1 (Yosemite, Retina display). Does
resizing the window cause them to pop back into existence?

tracernz, 2015-01-24:
I'm running 0.6.0 on Chrome 41, Yosemite, retina screen. It does switch to  The tabs disappear when I connect and reappear when I disconnect they reappear. Resizing the window doesn't make them come back. Even though they're not visible I can still click them and switch pages. I haven't had this issue until yesterday but I'm not exactly sure which version it began to happen unfortunately.

hydra, 2015-01-24:
Chrome 41 is beta version i think.
```
Version 40.0.2214.91 (64-bit)
 Google Chrome is up to date.
```
Is what my OSX Yosemite box says.

hydra, 2015-01-24:
p.s.  this is raised in the wrong issu tracker. it should be raised here: https://github.com/cleanflight/cleanflight-configurator/issues

tracernz, 2015-01-24:
I've just confirmed this is a problem with Chrome 41 (beta). Updated my Windows machine from 40 (tabs worked fine) to 41 and it now it's also broken.

thenickdude, 2015-01-25:
I see that on Chrome 41. Looks like it's fixed by adding the style
"overflow:hidden" to the <ul> that contains the tabs. (Adding
overflow:hidden causes an element's height to grow to contain any floated
elements that are contained inside it).

hydra, 2015-01-26:
this fix has been merged in to latest configurator development branch.

closed 2015-01-26 by hydra