cleanflight/cleanflight #423: Don't Arm ESCs Until Valid RC Input

tracernz, 2015-01-24:
It would be nice if the motor PWM outputs weren't started until either a valid RC signal is received or the motor test mode is enabled. I'd feel a lot more comfortable powering up on the test bench to make configuration changes this way.

hydra, 2015-01-24:
That's not a bad idea. I do wonder about general ESC compatibility though.

tracernz, 2015-01-24:
I know my helicopter flybarless controllers do this (that's where the idea came from ;-)).

ledvinap, 2015-01-25:
Most ESC will beep without PWM signal, so it would be quite annoying.

hydra, 2015-01-25:
@ledvinap yes, that occured to me too.  I thought about perhaps making this an option, but this also has the  potential to confuse users.  I'm really not sure about the benefits of this.

@tracernz i'm struggling to find a reason why you'd need this, since if the ESC's were powered up the RX would be too and depending on the RX you may get a valid input signal even without your TX on.

tricopterY, 2015-01-28:
As a user, I would very much prefer to have feature MOTOR_STOP enabled by default. Or, have a check - when Serial MSP/CLI is active, do not Write Motors. Not sure, how that would affect bluetooth comms and field PID/Rate tuning etc.

tracernz, 2015-01-28:
@ledvinap Hmmm, my BLHeli ESCs don't beep and neither do my Castle or Jeti ESCs on my helis.

@hydra It's just added safety as you know the motor definitely can't start at least without going through the arming sequence (and playing the arming tone) first.

hydra, 2015-02-20:
As an update to this, i was given an ESC recently that requires a PWM signal to be present immediately after power up.  If the signal is not present the ESC will not work.  Any PWM signal subsequently generated is always ignored by the ESC.  My assumptions were correct, ESC compatibility would break.

I don't think I want to add another setting which needs to be disabled by default that most people never will need or change.

I will close this issue and we can re-open it if other information and reasons are given.

closed 2015-02-20 by hydra