cleanflight/cleanflight #431: Blackbox Compatible with DashWare

andymac4182, 2015-01-27:
Hi,

Is it possible to get some example data from Blackbox. Was looking at using the data to overlay some videos using the DashWare software. http://www.dashware.net/how-it-works/ If that is possible it would be great.

thenickdude, 2015-01-27:
Sure, you can decode any of the logs from [this ticket](https://github.com/cleanflight/cleanflight/issues/381) using the [`blackbox_decode` tool](https://github.com/cleanflight/blackbox-tools) which will turn them into CSV files. Or if you like, here's one of mine already decoded to CSV:

http://s3.nicksherlock.com/LOG00548.01.Armattan%20226.Acro%20LOS.csv.zip

GPS data (in the form of latitude/longitude/altitude coordinates) can also be included in the CSV file, but I'm afraid I don't have an example of that to share.

andymac4182, 2015-01-27:
That is coming up as access denied for me.

That is okay thanks. I will get a data logger and give it a go.

closed 2015-01-27 by andymac4182

thenickdude, 2015-01-27:
Whoops, I mangled the URL, the correct URL is:

http://s3.nicksherlock.com/blackbox/LOG00548.01.Armattan%20226.Acro%20LOS.csv.zip