cleanflight/cleanflight #433: confgurator Backup button doesn't do anything.

cmanley, 2015-01-27:
I'm having this issue in the configurator v0.60.0.
My board is a naze32 with 1.6.0 firmware. 
The Restore button does work.
Two other things don't though (not sure if they're related): 
When I click on the tabs Ports or Configuration, then I only get "Waiting for data..." and packet errors.

B.t.w. I really appreciate your great work (cleanflight).

fiveangle, 2015-01-27:
Works for me and countless others on same HW+FW version.  It is highly unlikely this is a bug.  It would probably be best if you close this issue and use the forums for support.  If after investigating you find an actual bug, with a recipe to reproduce it, resubmit the defect.

-=dave

cmanley, 2015-01-27:
More info:
The issue occurs on Windows 7 64 bit, using various USB ports and hubs.
On my Win32 laptop, the issue does not occur. 
I traced it in Chrome inspector to some MSP callback not being called, but I didn't spend too much time on it yet.

@fiveangle I don't mean to be as rude as you, but what makes you expert on bug probability? Ever heard of OS/hardware specific differences such as number of ports and varying timeout periods, missing or incomplete exception/error handling, etc. It's definitely not a user error as if I clicked too hard on the button.

hydra, 2015-01-27:
Hi @cmanley fiveangle wasn't being rude.  It's hard to convey tone in messages such as these.  He is right however,the first port of call for problems is not the issue tracker, it's either the IRC channel or forums.  The issue tracker should be used to document bugs once the issue has been identified, giving as much detail and steps on how to re-produce the bug.

I will keep this open and mark it as blocked until you can give us a bit more information as right now i've got nothing to go on that would help track down the problem.

What happens if you use the 'defaults' command and then try the backup button.
What happens if you reflash and erase the chip?
What is the output of the console log when the configurator is run using the 'chrome developer tools'?
etc, etc.

closed 2015-01-27 by cmanley

hydra, 2015-01-27:
@cmanley did you fix the problem?

cmanley, 2015-01-27:
No

ledvinap, 2015-01-27:
I had probably same problem after updating to latest chrome on W7-64 - after pressing 'backup' nothing happened. other tabs did not load and packet error counter slowly incremented. 

But I can't  reproduce it again ...

fiveangle, 2015-01-27:
@cmanley - wasn't trying to be rude - no hard feelings.  Please update if you find a recipe to repro, as it looks to be pretty rare, whatever it is.

-=dave