cleanflight/cleanflight #434: differential ailerons on airplanes

klHaribo, 2015-01-27:
Is it possible to insert a cli variable for differentiated ailerons ?
for example aileron_differential 
= 100: aileron1 all the way up, aileron2 neutral
= -100: aileron1 neutral,aileron2 all the way down
= 0:  normal (default)

hydra, 2015-01-27:
can you point us at a good resource explaining what differential ailerons are and how they work?

klHaribo, 2015-01-28:
hi Dominic,

many thanks for your interest.

Short explaination:
Differentiated ailerons are intended to eliminate the adverse yaw . At 
the same time they improve the stall caracteristics.

I found two links, that explaines it very good:

http://www.rc-airplane-world.com/aileron-differential.html
http://www.skyhighhobby.com/what-is-aileron-differential.html

I nearly see differential ailerons on every airplanes.
Nearly every radio has the option for differential ailerons ;)

Excuse me for my bad english, The school hours are a bit longer ago .
  ;)

yours sincerely,

Dennis

universam1, 2015-02-12:
Hi, this would most probably be the solution for that problem:

http://www.multiwii.com/forum/viewtopic.php?f=7&t=6132

hydra, 2015-05-15:
I'm copying the comment from the forum thread for reference, not sure how much of it applies without reviewing the current code.

Proportional servo reflection

Postby universam » Thu Feb 12, 2015 11:08 am
In our club the discussion why MWII and Baseflight are currently not ready for serious fixed wing & flying wing usage is that it kind of a strange servo reflection. Honestly, in 30 years RC I haven't seen one single TX working like this. :?
As long as you keep servo middle at 1500 and max at 1000/2000 everything is fine, the problem is how MWII/Baseflight work when you change one / more of these settings: One side will be constrained and the other reflection wont be reached... This is a no-go for serious planes where you need asymetric servo reflection for flaps, ailerons and so forth

For better understanding see the illustration, left the current problem and right the result of the change.

Actually it is very easy to improve! With this changes MWII will work proportional and respect the max & min servo reflection. So a full extension of your stick will result in the max/min servo reflection.

When you think twice, you realize that now only min / mid / max is of interest, the servo rates dont make sense any more :o . But for compatibility I left that in calculation, as long as you keep it on 100 it really means 100%. A different value here could be also done, but I have no idea for an application.

The code changes simple!
Airplane mod, replace one line ~1336

Code: Select all
        for(i=3;i<7;i++) {
          // servo[i]  = ((int32_t)conf.servoConf[i].rate * servo[i])/100L;  // servo rates
        
        // propotional Servo patch
           // here we honor the servo rate and reversing! if rate is 100 we get full propotional path
           int32_t path = (int32_t)conf.servoConf[i].rate * servo[i];
               // depending on the side the left path to max/min is different so taking that in account for propotional path
           servo[i] = (path*(path>=0 ? conf.servoConf[i].max-conf.servoConf[i].middle:conf.servoConf[i].middle-conf.servoConf[i].min))/(100L*500L); 

          servo[i] += get_middle(i);
        }

Fixed Wing mod, replace ~1231 - this should be probably aligned to 2.4

Code: Select all
       for (i=3; i<=4; i++) {
        if (f.PASSTHRU_MODE) {    // do not use sensors for correction, simple 2 channel mixing
          servo[i] = (SERVODIR(i,1) * rcCommand[PITCH])*PITCHRATE + (SERVODIR(i,2) * rcCommand[ROLL])*ROLLRATE;
        } else {                  // use sensors to correct (gyro only or gyro+acc according to aux1/aux2 configuration
          servo[i] = (SERVODIR(i,1) * axisPID[PITCH])*PITCHRATE   + (SERVODIR(i,2) * axisPID[ROLL])*ROLLRATE;
        }
               // depending on the side the left path to max/min is different so taking that in account for propotional path
       servo[i] = (int32_t) servo[i]*(servo[i]>=0 ? conf.servoConf[i].max-conf.servoConf[i].middle:conf.servoConf[i].middle-conf.servoConf[i].min)/500;
            servo[i] += get_middle(i);
       }

Just for noting that, this code is currently being flown by several guys in our club with great success. I'm flying this already since months, and have no drawback. A sentence regarding the calculation time - I counted 37 more instructions in sum and the round trip could be a bit longer but I was unable to see any differences than before. So I can confirm it not realy time consuming.

I really hope this or something similar would go into master for the future.

Comments welcome!
Sam

[image]

jo930, 2015-12-20:
I can confirm that this is working.
There were just little adjustments necessary for the code to work with CF:

In cleanflight/src/main/flight/mixer.c I changed the code in line 735 - 738 from currently
```
    for (i = 0; i < MAX_SUPPORTED_SERVOS; i++) {
        servo[i] = ((int32_t)servoConf[i].rate * servo[i]) / 100L;
        servo[i] += determineServoMiddleOrForwardFromChannel(i);
    }
```
to
```
for (i = 0; i < MAX_SUPPORTED_SERVOS; i++) {
    int32_t path = (int32_t)servoConf[i].rate * servo[i]; 
    servo[i] = (path * (path>=0 ? servoConf[i].max - servoConf[i].middle : servoConf[i].middle - servoConf[i].min)) / (100L * 500L);  
    servo[i] += determineServoMiddleOrForwardFromChannel(i);
}
```
Had a successful test flight today with that code on my airplane.

Johannes

universam1, 2015-12-20:
Good job!
Thanks for porting! :+1:

closed 2017-02-25 by hydra

glideher, 2018-04-07:
when it comes to differential i keep things simple, always set my up elevators 5% more than down, from there i fly and adjust as needed, its a feel thing for me, i like to feel every adjustment i make and re adjust accordingly, trial and error,,with 3D airplanes not so much differential is needed due to the symmetrical airfoil, great discussion by the way :)
another helpful thread here http://www.multiwii.com/forum/viewtopic.php?f=7&t=2456&p=61468#p61468

cheers

[dlg gliders](https://www.microbirds.com)