cleanflight/cleanflight #435: GPS Tab issue

waslater, 2015-01-27:
I currently have a Crius ublox neo-6m (no integrated compass) connected to my naze board. This one: (http://www.massiverc.com/PrestaShop/en/409-neo-6-v31-gps-neo-6m-module.html) It appears to work, but I can never get GPS hold mode to enable. Wjhen everything is powered on outside, the GPS tab shows a fix and an accurate latitude and longitude (I don't know how), but nothing appears in the satellite strength indicator box. I know the GPS hold mode is not engaging because the 5 satellites are not present. 

Does anyone know how or why cleanflight is detecting the presence of my gps, giving me a lat/lon, but not idicating individial signal strengths from any of the satellites?

hydra, 2015-01-27:
Yes, it's probably because the SVINFO packet is not being received by the FC.  Please either use the auto configuration cli option or configure your gps unit manually as per the documentation.

The issue is not in the configurator.  There is no issue retreiving the gps signal strength from the fc, just that the fc has not receivd signal strength information.

closing the issue, please re-open if you are are sure your configuration is correct after verifying it.

closed 2015-01-27 by hydra

waslater, 2015-01-27:
Thanks for the tips, I'll try the auto configuration again.