cleanflight/cleanflight #436: Blackbox SDFAT benchmark repository

fiveangle, 2015-01-27:
This is a repository for SPI interface benchmarks of various microSD cards using the SdFat "bench" sketch uploaded to your OpenLog device.

-Install the SdFat libraries in your Arduino IDE
-Upload the "bench" sketch to your OpenLog device
-Open Serial Port Monitor of your Arduino IDE at 9600 baud
-Instructions will be in SPM on how to run test
-Copy test output from SPM
-Reply to this thread with at least this info:
--Report the product you ran the benchmark on (e.g. Witespy GPS Logger) with URL if handy
--Report the SD Card (e.g. Sandisk 4GB mSDHC Class-10 - came installed with Witespy GPS Logger purchased Jan 2015)
--Test output

SdFat suite:
https://github.com/greiman/SdFat

SdFat "bench" utility:
https://github.com/greiman/SdFat/tree/master/SdFat/examples/bench

SDFormatter (official SD Foundation utility that can align flash erase boundaries with file
system structures which can sometimes reduce write latency and file system overhead).

Thanks !

-=dave

fiveangle, 2015-01-28:
Witespy GPS Logger
Sandisk 4GB mSDHC Class-10 (came installed in Witespy GPS Logger) - out-of-box condition

    Free RAM: 615
    Type is FAT32
    Card size: 4.03 GB (GB = 1E9 bytes)

    Manufacturer ID: 0X6F
    OEM ID:
    Product: SMI  
    Version: 0.0
    Serial number: 0X0
    Manufacturing date: 3/2014

    File size 5 MB
    Buffer size 512 bytes
    Starting write test, please wait.
    write speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    243.58,18456,1916,2095
    228.85,286064,1920,2229
    231.19,286108,1916,2208
    231.16,286248,1916,2208
    231.27,286288,1916,2207
    230.02,286136,1860,2219
    232.76,286020,1856,2193
    232.49,286248,1856,2196
    232.77,285980,1856,2193
    232.44,286004,1856,2196

    Starting read test, please wait.

    read speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    459.07,2420,1072,1109
    459.07,2420,1072,1109
    459.07,2420,1072,1109
    459.11,2420,1072,1109
    459.11,2424,1072,1109

    Done

-=dave

thenickdude, 2015-01-28:
I'll note that tests are basically worthless unless you do a full erase of the card with sdFormatter first. I got a ten-fold reduction in maximum write latency by erasing my well used SanDisk card (from 95ms to 10.6ms).

Here are some benchmarks from the author of SDFat himself:

http://forum.arduino.cc/index.php?topic=258667.0

And tests I've performed:

SanDisk Extreme 16GB:
```
Type is FAT32
Card size: 15.93 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SE16G
Version: 8.0
Serial number: 0XA4E3C26
Manufacturing date: 5/2014

File size 5 MB
Buffer size 512 bytes

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
284.82,25620,1512,1791
295.47,10644,1532,1727
295.70,10652,1600,1725
295.35,10636,1600,1727
295.28,22104,1540,1728
295.49,10640,1600,1726
295.52,10640,1596,1726
295.52,10640,1536,1726
295.49,10628,1600,1726
295.26,22008,1600,1728

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
495.85,2020,1008,1026
495.95,2028,1008,1026
495.90,2020,1008,1026
495.90,2020,1008,1026
495.95,2024,1008,1026
```

Transcend 16GB Class 10 UHS-I:
```
Type is FAT32
Card size: 15.93 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SL16G
Version: 8.0
Serial number: 0X52D5ED0E
Manufacturing date: 8/2014

File size 5 MB
Buffer size 512 bytes

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
273.27,44620,1556,1867
272.43,59888,1552,1873
271.99,89828,1552,1876
273.43,44352,1552,1866
271.75,59868,1552,1878
272.20,89508,1552,1875
273.27,44372,1552,1867
272.22,59852,1552,1874
272.20,89544,1552,1875
273.16,44296,1552,1868

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
480.88,2116,1048,1059
480.92,2116,1048,1058
480.92,2120,1048,1058
480.92,2112,1048,1059
480.92,2116,1048,1058
```

fiveangle, 2015-01-28:
Witespy GPS Logger
Monster Digital Safari Series 8GB mHDHC Class-10 (purchased approx Sept 2014)
Freshly long-formatted with SDFormatter

    Free RAM: 615
    Type is FAT32
    Card size: 7.97 GB (GB = 1E9 bytes)

    Manufacturer ID: 0X27
    OEM ID: PH
    Product: SD08G
    Version: 3.0
    Serial number: 0XD1F89A01
    Manufacturing date: 6/2013

    File size 5 MB
    Buffer size 512 bytes
    Starting write test, please wait.

    write speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    138.80,184644,2552,3668
    138.88,308860,2676,3666
    138.10,308668,2668,3686
    138.97,302376,2672,3662
    138.33,158820,2672,3680
    138.79,302316,2664,3668
    137.80,302992,2664,3694
    139.00,309432,2668,3662
    136.18,310204,2664,3745
    138.81,239040,2668,3667

    Starting read test, please wait.

    read speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    368.14,2712,1244,1385
    368.16,2704,1240,1385
    368.27,2708,1240,1384
    368.38,2708,1240,1384
    368.46,2704,1244,1384

    Done

-=dave

fiveangle, 2015-01-28:
Witespy GPS Logger
Patriot 16GB mHDHC Class-10 (purchased approx Aug 2014)
Freshly long-formatted with SDFormatter

    Free RAM: 615
    Type is FAT32
    Card size: 15.72 GB (GB = 1E9 bytes)

    Manufacturer ID: 0X27
    OEM ID: PH
    Product: SD16G
    Version: 3.0
    Serial number: 0X48C1967C
    Manufacturing date: 3/2014

    File size 5 MB
    Buffer size 512 bytes
    Starting write test, please wait.

    write speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    187.85,114804,2004,2718
    184.39,103152,1996,2769
    187.06,104252,2000,2729
    182.60,107740,2008,2797
    187.20,104832,2004,2727
    184.35,106212,2000,2770
    186.92,103024,2004,2731
    184.24,103048,2008,2771
    186.27,103228,2004,2741
    183.40,105372,2004,2783

    Starting read test, please wait.

    read speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    434.60,2260,1032,1172
    434.68,2260,1032,1172
    434.79,2268,1032,1172
    434.75,2260,1032,1171
    434.75,2260,1032,1172

    Done

-=dave

fiveangle, 2015-01-28:
@thenickdude - Sandisk has discontinued all but the 64GB extreme (leaving the less-capable Evo series to fill that gap) so unless you want to plunk down that much cash, keep your 16GB Extreme  coveted :)

I also felt that benchmarks where a card is not freshly formatted could be useful since (at this point) we don't know for certain how "used" a card needs to get before it's streaming IO speed deteriorates.  It may be that we need to do the long drawn out format before every flying day, or maybe just once every few months (given the relatively little amount that gets logged, compared to the size of the card).  The benchmarks should tell us either way.

Yes, it is precisely that dead post in the Arduino forums (which you resurrected :+1: ) and the wildly varying results I obtained on the cards I had laying around, that prompted me to start this repo.

With the latest autotune logging code pushing far beyond the logging limits already, I have a feeling stumbling across the next cheap, fast-SPI mSDHC card is going to be a hurdle we need to jump before the blackbox logging is able to become our ubiquitous troubleshooting tool.

Anyway, and I thought it was interesting :)

Thanks for posting up your goods !  I have a few more to come...

-=dave

thenickdude, 2015-01-28:
Thanks for the heads-up, I put in an order for three more SanDisk Extreme 16GB!

We can use slower cards, but it requires a substantial amount of memory available to buffer while the card hangs. At 14400 bytes/second, a maximum SD card write latency of 150ms requires 2KB of buffer RAM. On the fancier STM32 chips, this is quite reasonable, and we could add a MicroSD card slot to the board and have Cleanflight perform the writes itself. I'm also looking in to logging with a Teensy 3.1, which has 64kB of RAM and so could log on the most dire cards, instead of the OpenLog.

fiveangle, 2015-01-29:
Witespy GPS Logger
Generic 1GB SD that came in Taranis Plus (purchased approx Aug 2014)
With 22MB of OpenTX system and sound files installed on it and no SD-card logging enabled in TX, so otherwise as-received

    Free RAM: 615
    Type is FAT16
    Card size: 1.02 GB (GB = 1E9 bytes)

    Manufacturer ID: 0X36
    OEM ID: AO
    Product:      
    Version: 0.0
    Serial number: 0XB10108FF
    Manufacturing date: 2/2009

    File size 5 MB
    Buffer size 512 bytes
    Starting write test, please wait.

    write speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    126.77,150000,2228,4018
    134.10,151008,2228,3797
    134.38,150408,2232,3789
    134.46,157072,2232,3787
    134.51,157356,2216,3786
    134.51,156116,2208,3786
    134.63,156676,2228,3782
    134.42,153524,2224,3789
    134.65,152368,2216,3782
    134.51,152488,2232,3786

    Starting read test, please wait.

    read speed and latency
    speed,max,min,avg
    KB/Sec,usec,usec,usec
    471.76,2160,1036,1079
    471.76,2160,1036,1079
    471.80,2160,1036,1079
    471.80,2164,1036,1079
    471.76,2164,1036,1079

    Done

Pretty pathetic, but at 150ms, still nearly half the max latency of the Monster Digital 8GB Class-10

-=dave

neurorotors, 2015-02-20:
I just tested Teensy 3.1 with Sandisk microsd 16GB.
It has been intensively used in raspberry pi over an year.
latency is pretty high overall. 
With the lager buffer, writing performance get drastically increased.

Type any character to start
Free RAM: 57247
Type is FAT32
Card size: 16.04 GB (GB = 1E9 bytes)

Manufacturer ID: 0X2
OEM ID: TM
Product: SD16G
Version: 4.1
Serial number: 0X4BF528D3
Manufacturing date: 10/2009

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
76.20,185037,697,6717
66.67,193671,3063,7677
69.65,191362,3057,7349
70.43,190145,3054,7267
66.30,191412,3058,7720
68.10,190816,3057,7516
69.35,189612,3051,7380
65.27,191968,3056,7841
66.96,192396,3056,7644
68.28,191606,3053,7496

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1102.95,3047,416,463
1103.93,862,416,462
1103.68,862,416,463
1103.20,862,415,463
1103.20,862,416,463

Done
Type any character to start

Use a freshly formatted SD for best performance.

Type any character to start
Free RAM: 49055
Type is FAT32
Card size: 16.04 GB (GB = 1E9 bytes)

Manufacturer ID: 0X2
OEM ID: TM
Product: SD16G
Version: 4.1
Serial number: 0X4BF528D3
Manufacturing date: 10/2009

File size 10 MB
Buffer size 8192 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1323.21,442371,4161,6181
1492.35,71517,4161,5483
1507.88,72266,4160,5427
1489.90,66886,4160,5493
1499.96,60551,4161,5455

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2373.93,6087,3416,3449
2375.62,3897,3418,3447
2375.06,3895,3418,3447
2375.62,3898,3417,3447
2375.06,3896,3418,3447

Done

thenickdude, 2015-02-20:
Yeah, if you have a Teensy you should be able to edit the buffer size in
OpenLog to 8kB or more and get flawless logging even at double the regular
115200 baud rate. You can probably up the baud rate of the ports on both
sides as well.

>

hydra, 2015-05-15:
This is a misuse of the issue tracker since there is nothing to do.  Though yes, there is no other place right now.  Closing anyway, people can still reference this issue and add comments/logs as required.

closed 2017-02-25 by hydra