cleanflight/cleanflight #438: Log parameters to Blackbox

olelille, 2015-01-28:
Hi,
It would have been nice to log PID settings and other parameters to blackBox. On the way, you get an overview of the options you have used for the various logs. Use a separate log file for the parameters with time stamp.

thenickdude, 2015-01-28:
Yeah, I think it would be useful to log at least the PID settings and the pid_controller, did you have any others in mind?

olelille, 2015-01-28:
looptime, gyro_lpf and everything else that have impact on the regulator.

olelille, 2015-01-28:
I just think that the log files are useless without knowing the system settings.

fiveangle, 2015-01-28:
I bit like using a bazooka to swat a gnat, but how about just stream a full config dump at startup ?

-=dave

andymac4182, 2015-01-28:
Would it be possible to log the whole dump set of commands? That would mean that you could recreate someones config fairly closely.

thenickdude, 2015-01-28:
olelille: looptime is measured, so you can see your actual achieved looptime rather than what it was set to, which is more useful anyway. gyro_lpf might be useful.

The full (text) config dump can't be used because it's so massive. Looks like my config dump would take a full second to stream out over the Blackbox port. Presently there's no way to produce it line-by-line on demand, so the next best thing would be to buffer it in memory and send it out, but that would require far more memory than is available.

fiveangle, 2015-01-28:
walk through every value in clivalue_t with cliget ?

-=dave

thenickdude, 2015-01-28:
It's not really going to be much smaller than the text dump since field
names and types still need to be included.

olelille, 2015-01-28:
What about saving the parameters after the flight when disarming the quad ? 
When arming the drone you want to start logging immediate to be able to sync the data to the buzzer beep.  After flight you at least have some seconds before removing the power.

thenickdude, 2015-01-28:
Hm, that is an interesting idea actually!

hydra, 2015-05-15:
Ok, I propose that is is blocked until configuration storage is reworked.  At that point it should be possible to just dump the configuration storage block of eeprom into a file/log. Much simpler.

thenickdude, 2015-05-15:
I don't think logging the flash is helpful because it means that every future `blackbox_decode` needs to keep the precise details about every previous Cleanflight release. I was thinking of exporting the CLI's variable reference and using that.

hydra, 2015-05-15:
See #347

hydra, 2015-05-15:
@thenickdude that's the whole point, we DON'T want each blackbox log to know about the format of the storage and all data for every version! we just the types of values and the values themselves and how to retrieve them and print them out.

thenickdude, 2015-05-15:
Yes, that was my point.

waltr1, 2015-06-27:
I would also like to see at least all the PIDs in the LOG header. Keeping paper notes as to what you changed and then guessing at which log file these correspond to is tedious.
It does dump some values that are not necessary (minthrottle, maxthrottle). These would NOT be changed while Tuning whereas the PIDs and now the newer "filter" variables introduced by Boris.
.

thenickdude, 2015-06-28:
Min/maxthrottle are logged because they're required for working out what
percentage value is being transmitted to the motors. The craft motor
display wouldn't exist without them.

waltr1, 2015-06-28:
Ok, thanks for the explanation of min/max throttle.

Still would like all the PIDs logged.

digitalentity, 2015-06-28:
Adding PID logging to blackbox header is trivial: add the required values to the `blackboxWriteSysinfo`. I might post a PR for this in a few days - in need of recording PID values to blackbox myself. Very useful for debugging stuff.

joshuabardwell, 2015-06-28:
>I would also like to see at least all the PIDs in the LOG header. Keeping paper notes as to what you changed and then guessing at which log file these correspond to is tedious.

Second this. Even if it took an extra second after arming to start logging... you could delay the sync beep, perhaps, and/or I think we can stand to lose one second of data at the start of the flight.

thenickdude, 2015-06-28:
I'm planning to dump all cli variables including the PIDs since there seems
to be demand for a range of variables. I've got it half complete already
but it's blocked waiting for other PRs to merge first.

closed 2017-02-25 by hydra