cleanflight/cleanflight #439: Receiver Failsafe mode

Viald, 2015-01-28:
It seems that CF (neither BF) doesn't handle correctly the signal sent by the receiver FS-R9B when the transmitter failed. See this video for explanations and details. http://vimeo.com/117964930
Thanks

fiveangle, 2015-01-28:
You're using PWM and all your failsafe values are mid-stick. You need to configure FS on the reciver correctly (or switch to PPM input if possible where FrSky will send an active fail safe indicator).

More info: http://www.rcgroups.com/forums/showthread.php?t=2249574&page=56

-=dave

Viald, 2015-01-28:
No I'm using PPM. As you can see the fail safe indicator seems to be sent as the board is flashing red with the sound alert. As I know it's not possible to configure FS on this kind of receiver. 
FS values aren't sent by the receiver. Have a look to the same test with Baseflight in the second part of the video, you'll see that with BS the FS values are different.

fiveangle, 2015-01-29:
No, I said you are using PWM (discreet inputs). I do not have a PWM setup to test, but would assume PWM signals for the first 4 ch would go to zero if RX gets no valid signal. Yours appear to have mid-stick input by CFC monitor. But I may not be understanding correctly. There is also a known problem for RXs that don't drop PWM signal when TX signal is lost, preventing FS. You say the beeping is occurring indicating FS should be active, so perhaps someone else with more experience with PWM input will comment. Good luck,

-=dave

Viald, 2015-01-29:
Thank you @fiveangle 
The ticket you pointed me has quite same issue. I ask Philippe to test his patch.

hydra, 2015-01-30:
Please test firmware 1.7.0 and update this issue once you have tested it.

If you still have the problem please re-open this issue.

closed 2015-01-30 by hydra

Viald, 2015-01-30:
1.7.0 FS works like a charm. Thank you.