cleanflight/cleanflight #440: Set UART1 with TELEMETRY

tony056, 2015-01-28:
Can I set TELEMETRY(before armed) to UART1?
Because I want to control my copter with my smartphone via bluetooth. (app: EZ-GUI)
Is there any advanced configuration or I can use default setting to connect with EZ-GUI?

fiveangle, 2015-01-28:
I'm not sure what you're asking, but you may want to reach out to the community for help with configuration such as at RCGroups: http://www.rcgroups.com/forums/showthread.php?t=2249574

This is primarily for software defect tracking (and enhancement requests).

-=dave

tony056, 2015-01-29:
I want to control my copter with my smartphone(EZ-GUI) via bluetooth. Now, I have to arm it with rc receiver, but I don't want to plug a rc receiver on my copter. Is it possible?

hydra, 2015-01-29:
Please the forums or IRC for assistance, not the issue tracker.  I'm sure you'll find help if you ask in the right place.

closed 2015-01-29 by hydra