cleanflight/cleanflight #442: Motors start spinning when quitting CLI

avivik, 2015-01-28:
Quad is connected with USB cable to the laptop, and also connected to a battery.
I was viewing parameters on CLI (didn't change anything), and when I hit the 'disconnect' button in the configurator, the motors started spinning for 2 seconds and then stopped (because the FC was rebooting as a result of quitting the CLI).
I repeated the scenario, and while this phenomenon doesn't happen all the time, it happens ~80% of the time.
This doesn't happen when quitting the CLI the "right" way, by typing 'exit' or 'save'.

Naze32 v1.6.0
Configurator 0.6.0
Windows 7 Pro 64-bit

SlyMike, 2015-01-28:
I had similar issues #385

hydra, 2015-01-29:
Duplicate of #123

hydra, 2015-01-29:
Can you update on #123 with details of which ESCs you are using?

closed 2015-01-29 by hydra