cleanflight/cleanflight #443: Can't open LED tab with FW v1.7.0 NAZE 2015-1-27 23:21 (rc)

SlyMike, 2015-01-28:
Flashed above FW, full chip erase.
Running - OS: Windows, Chrome: 40.0.2214.93, Configurator: 0.60.0 - DEFAULT SETTINGS

Can't open LED tab, just getting packet errors and "waiting for data" message.

Added following with CLI:

defaults
save

(reboot)

SET ALIGN_BOARD_YAW=90
set align_mag = 1

set serial_port_2_scenario = 8 
set msp_baudrate = 115200

feature RX_PPM
feature VBAT
feature FAILSAFE
feature SONAR
feature TELEMETRY
feature LED_STRIP

led 0 0,15:W:IWA:0
led 1 1,15:SW:IWA:0
led 2 2,15:S:FT:0
led 3 3,15:S:FT:0
led 4 12,15:S:FT:0
led 5 13,15:S:FT:0
led 6 14,15:ES:IWA:0
led 7 15,15:E:IWA:0
led 8 15,0:E:IWA:0
led 9 14,0:NE:IWA:0
led 10 13,0:N:F:0
led 11 12,0:N:F:0
led 12 3,0:N:F:0
led 13 2,0:N:F:0
led 14 1,0:NW:IWA:0
led 15 0,0:W:IWA:0
led 16 0,0:::
led 17 0,0:::
led 18 0,0:::
led 19 0,0:::
led 20 0,0:::
led 21 0,0:::
led 22 0,0:::
led 23 0,0:::
led 24 0,0:::
led 25 0,0:::
led 26 0,0:::
led 27 0,0:::
led 28 0,0:::
led 29 0,0:::
led 30 0,0:::
led 31 0,0:::

save

(reboot)

LED settings are applied and functional but CLI tab still fails to work.

Also having to disconnect to leave CLI.

newts, 2015-01-28:
I had this same problem (NAZE) and verified that it is resolved in the dev version of the code.

fiveangle, 2015-01-28:
Configurator has its own project so it's best to open issues like this against that component.

This is likely a dupe of https://github.com/cleanflight/cleanflight-configurator/issues/70

-=dave

SlyMike, 2015-01-28:
OK, cheers.

Just realised that the FW was updated but not the configurator.
I'm sure this will be resolved when the configurator is updated to accept the changes in the new FW.

closed 2015-01-28 by SlyMike