cleanflight/cleanflight #450: Sparky board soft-bricked after flashing cleanflight >= 1.6

emdzej, 2015-01-29:
After flashing Sparky with cleanflight 1.6 or above OS dosn't recognize the board in VCP mode anymore.
Tried different flash methods:
- dfu-util under OSX (including mass-erase)
- defuse under Win
Binaries taken either from releases page (hex converted using filemgr) or self-compiled. Every time the result is the same. OS reports bad USB descriptor - under Windows, or enumeration failed under OSX.
Flashing 1.5 back resolves the issue

hydra, 2015-01-29:
rev1 or rev2?

hydra, 2015-01-29:
Both of my two sparkys are working with latest 1.7.0 release candidate.

emdzej, 2015-01-29:
TauLabs GCS is reporting rev 2. I know that's weird, but tried many approaches with no luck, unless I'm missing something obvious.

hydra, 2015-01-30:
please can identify the revision visually - the differences are here:

https://github.com/TauLabs/TauLabs/wiki/Sparky

some newer binaries have been built which corrected a problem with sbus port selection.  please can you try with the latest release candidate?

hydra, 2015-01-30:
Might be related to #452

emdzej, 2015-01-30:
Small update. So it's definitely an earlier revision (with 4 pin receiver port). I have downloaded the latest compiled binary from the releases and it worked this time. Then I've tried to compile the binary and flash it and it did not. (No luck with 1.6 though) 

The 1.7 binary downlaoded and 1.7 compiled differ - if this helps. Maybe there's something wrong with my toolchain, although no compilation errors occur.

hydra, 2015-02-03:
ok, unable to replicate the problem here.  yes, every time you compile it you will get a different file since the date/time are embedded in the binary.  i'm using gcc 4.8, not 4.9.

Closing since latest release version works.

GCC 4.9 issues will be addressed sepatately but can reference this issue if needed.

closed 2015-02-03 by hydra