cleanflight/cleanflight #451: PID Controller 1 P_level default value needs changed.

dave-pitman, 2015-01-29:
On PID Controller 1,  the default value for Level_P is 90 which from all accounts is very unstable, and is uncontrollable.  Lowering the default value to something like 20 would make it workable.

If someone knows where in the source this default is located, please post.  Or if easier, just provide a commit with the solution.  Thx.

dave-pitman, 2015-01-30:
Apparently these defaults cannot easily be changed.  I submitted a pull request to make a note that the user should adjust this value before attempting Angle mode using PID Cont. 2

hydra, 2015-02-06:
Closing.  PR for documentatio merged. Can't really do anything about this right now.  Further suggestions on how to fix welcome though.  

A longer term goal is to allow the configurator to suggest defaults.

closed 2015-02-06 by hydra