cleanflight/cleanflight #452: CC3D not connect version v1.7.0

Piradoxlanieve, 2015-01-30:
I have the CC3D controller with version v1.6.0 connects well but with version v1.7.0 does not connect and I have to flash from flash loader demostrator. Sorry my bad english

hydra, 2015-01-30:
please can you try flashing with a full chip erase and report back.

Piradoxlanieve, 2015-01-30:
i try "global erase" cc3d.hex works via  USB VCP and try flash with cc3d.bin not works uart and vcp. Sorry my english

hydra, 2015-01-30:
Please confirm:

A) After Flashing .hex the USB vcp works
B) after flashing .bin using GCS vcp does NOT work.

Piradoxlanieve, 2015-01-30:
A)yes work OK. B)(not connect port 10second and close port automatic) mini USB and fdti main port not work with flash .bin and led blue not work only led green

hydra, 2015-02-03:
A workaround has been added to 1.7.0 so that it's possible to use the either VCP or UART1 for MSP on the CC3D.

Root cause still unknown, probably something in the openpilot bootloader.

lamnesique, 2015-02-03:
hi, i'm in the same situation : i flash my cc3d with 1.7rc release all work fine can connect configurator via usbport.... when i want to flash the 1.7 stable via usb the flasher return error can't connect to bootloader...
so i try to connect via mainport...and it wont work can't connect to the board, i try via gcs and nothing work to and i still can't connect via stmflashloader with and without bootloader pin short ......the board still work, i can tune it via usb but can't do upgrade firmware......i wish that my explaination are clear..... thanks for your work

lamnesique, 2015-02-03:
I auto answer, if any person have the same pb you can't upgrade firmware via usb you also can by uart1 but you have to don't forget to configure uart1 with msp option before and save after that you upgrade without problemes and connect via both via usb and uart1.....it work fine for me....

hydra, 2015-02-03:
@lamnesique were you using the openpilot boot loader and gcs to flash?

lamnesique, 2015-02-03:
no at all.....in cleanflight i connect via usb (because just this work) and i go to port tab and i saw that usb vcp was set on msp,cli,gpspassthrough and all ports none...so i just set uart1 (mainport) with msp save and after that i was able to flash via the flashfirmware utility.....i note that you can add a scenario for port because if we want msp and cli on usb we can't set a scenario on uart1 with msp and gpspassthrough to have osd and gpspassthrough for it.....the best scenario for me : usb set in msp and cli just that and uart1 set in msp and gpspassthrough (optionnaly cli).....i don't know if my ideas are clear for you...i'm still here if it's necessary

hydra, 2015-02-04:
People have reported that upgrading your USB drivers can help.  Closing this issue, unable to replicate here. If after updating your vcp drivers you still have issues please reopen and provide further details.

closed 2015-02-04 by hydra

digitalentity, 2015-02-21:
@hydra, I can confirm that 1.7.2 @ CC3D is unable to communicate via USB. Updating VCP drivers does not help, device is simply not recognized by OS.

digitalentity, 2015-02-21:
An update. USB work fine on 1.7.1 and 1.7.2 (my bad, flashed wrong firmware). However USB does not work on latest development firmware. I've just checked out the latest master brach, compiled it, flahed and *bang*! No USB.

digitalentity, 2015-02-21:
Another update. This seem to be a compiler issue. When building with GCC ARM 4.9-2014-q4 usb vcp does not work. When building with 4.8-2014-q2 everything is fine.

thenickdude, 2015-02-21:
GCC 4.9 is known to produce flaky builds on Sparky too (noted in the [Linux docs here](https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Ubuntu.md)).

digitalentity, 2015-02-21:
@thenickdude, thanks for noting this. I am building CF in windows, so I never looked to the linux-related docs. May be we should edit the 'building in windows' docs and explicitly specify that 4.9 might provide faulty builds.

CopterFail, 2015-05-10:
typing "lsusb" in the linux console shows the problem:

with CLEANFLIGHT_CC3D.BIN:
Bus 003 Device 009: ID 20a0:415b Clay Logic CopterControl

with CLEANFLIGHT_CC3D.HEX:
Bus 003 Device 013: ID 0483:5740 STMicroelectronics STM32F407

The ID is still the openpilot id, no tty device is generated. The problem appeared after an updated openpilot gcs was installed - not directly a cleanflight problem. Does anybody know a workarount?

dascons, 2015-05-10:
Have you tried plugging a battery in before connecting the usb cord?

CopterFail, 2015-05-11:
Ok, in case the cc3d is powered before it is connected to USB the ID is for HEX and BIN:
ID 0483:5740 STMicroelectronics STM32F407. It is recognized by the configurator :)
So the problem has nothing to do with the openpilot gcs installation, but with the openpilot bootloader (it is the first started program on the cc3d and is identified by the usb enumeration)
The cc3d has to be replugged after each "save&restart"? 
Thank you for the help.