cleanflight/cleanflight #458: LED Strip - configuration screen not loading

JohnieBraaf, 2015-01-30:
I'm experiencing the following issue on Sparky flashed to 1.7:

The configuration manager for the LED string displays message: Waiting for data...
Nothing happens after that, I need to disconnect to make the UI responsive again.

hydra, 2015-01-30:
old configurator works with old firmware, new configurator works with new firmware.

what combination are you using that doesn't work?

Did you try 1.7 with 0.61.0 ?

hydra, 2015-01-30:
p.s. release notes of latest configurtor mention this situation.

JohnieBraaf, 2015-01-30:
Thnx, I was using 0.60.0 which I got as a Chrome extension.
I didn't see version 0.60.0 in cleanflight-configurator(-development) and master , so I've taken it from the development branch now. 
That version works.

closed 2015-01-30 by JohnieBraaf