cleanflight/cleanflight #459: feature current meter not saving...

handsomejackuk, 2015-01-30:
Hi have tried enabling current meter in CLI, and in the configuration gui tab, i have set all the offsets and scales and these appear to be ok. I am using the virtual current sensor sensor 2 is there any issues or resaon why the feature will not enable...?

 also saving before i leave the cli tab...

SlyMike, 2015-01-30:
Did you power cycle your copter? Pull the battery and usb lead?
Just tried on my NAZE32 and worked first time.

feature CURRENT_METER
set current_meter_type=2
set current_meter_scale = xxx
set current_meter_offset = yyy
set battery_capacity=zzz

handsomejackuk, 2015-01-30:
no i didnt do that but i will try once my batteries have charged... my scale is 186, and offset 41 i got 0.41 amps at minimum, and 30.22 amps at full throttle..thanks

hydra, 2015-01-30:
perhaps some other feature you have enabled is causing a conflict.

though, perhaps actually this is an issue since you're using the virtual one there should be no conflict.

this needs fixing.

handsomejackuk, 2015-01-30:
got sonar, and vbat, pwm receiver i did disable sonar. but still the same about to retest...

SlyMike, 2015-01-30:
Just wanted to point out that I had the following enabled:

feature RX_PPM
feature VBAT
feature FAILSAFE
feature SONAR
feature TELEMETRY
feature LED_STRIP
feature CURRENT_METER

But the only way I could verify that the feature was active was by connecting UART2 to a TTL to USB board via USB OTG cable and reading raw data with EZ-GUI as the configurator 0.61.0 (default UART / USB cable) seemed to lock up whenever I armed my copter via Tx.... But did resume comms once disarmed.

Should that be posted as a separate issue or is it a known condition for this version?

hydra, 2015-02-03:
This should be fixed. Please reopen if there's still an issue.

closed 2015-02-03 by hydra