cleanflight/cleanflight #462: Firmware v1.7.0 - Connected OLED display prevents Naze32 from proper bootup

snoekieboe, 2015-02-02:
Hi,

After updating the Naze32 firmware to version 1.7.0 RC the board no longer boots up propperly if a OLED display (I2C IIC Serial 128 x 64 OLED LCD) is connected.

Disconnecting the OLED display makes the board boot and function normally.
Downgrading to version 1.6.0 solves the issue and the Naze board is able to boot and function normally WITH the OLED display attached.

thenickdude, 2015-02-02:
Do both versions have feature DISPLAY enabled?

snoekieboe, 2015-02-02:
Yes, both versions have feature DISPLAY enabled.
What i forgot to mention:

When booting V1.7.0 the OLED display briefly displays "Cleanflight 1.7.0" at the top line and then the system halts...

hydra, 2015-02-06:
Still enable to replicate this, also tested fine on new F3 based prototype boards with no issue.

Will close this issue, if you can provide more details we can re-open and look into it further.

closed 2015-02-06 by hydra

snoekieboe, 2015-02-09:
FYI,

After replacing my board for a new one the problem does not exist anymore.

Regards,

Roel