cleanflight/cleanflight #465: Enabling softserial ports one by one

SupremeFPV, 2015-02-03:
Is there a possibility to add support for enabling the soft serial ports one by one? 
I am only using one of the softserial ports, but I still want to use current sensor.

Are there some hardware limitations that prevent this from happening or will the current sensor work even if i have softserial enabled?

hydra, 2015-02-03:
there's no hardware conflict, marking this as a todo for 1.8

pinkywafer, 2015-02-04:
Is there any reason LED has to be on pin 5, And Current on pin 8 on the naze board? If they were on 7+8 or 5+6 respectively, you could then still use both LED and current, AND one softserial with this enhancement? 
Jenny

hydra, 2015-02-14:
yes, timer to pin mapping is the reason.

P-I-Engineer, 2015-03-09:
this is great, this what i was looking for.

i'm assuming this would allow me to run osd on uart 1 and frsky telem on soft serial and still have a current sensor hooked up.

thanks for all the great work.

snoekieboe, 2015-03-19:
Did this feature make it into 1.8.0? It seems i still cannot make use of LED and Softserial at the same time?

hydra, 2015-03-19:
Not possible to use led and softserial at the same to.e on the naze.  Is possible to use current meter and softserial since current meter does not use a hardware timer. This ticket will be updated when the latter is done.

dkisselev, 2015-03-26:
@hydra is there another platform that has more HW timers? Or would it be possible to give LED the option of using a software timer? Looking to use LEDs with always-on telemetry (which requires softserial)

thenickdude, 2015-03-26:
You can use LEDs with always-on telemetry without using softserial. Just put telemetry on UART2.

dkisselev, 2015-03-26:
Ahh, my bad, I was going off of this (outdated, especially for 1.8) video: https://www.youtube.com/watch?v=dahGlc3Tf40, but just realized that softserial was needed in that case to handle the GPS as well as telemetry/LED's.

I guess to expand on the question, would it be possible to do always on Telemetry + GPS + LED's? I guess the solution would be to attempt to migrate either softserial or LED's off of a hardware timer, but is that possible (CPU usage-wise?

thenickdude, 2015-03-26:
I think pull request #177 attempts to use one hardware timer to serve several systems (combine timers). 

Otherwise you could use an F3 based system like the Sparky or Hydra's in-development flight controller, which support 3 hardware serial ports and an additional USB port (so you could have always-on telemetry and GPS on 2 of the hardware ports, CLI/MSP using the USB port, and not use softserial at all).

ledvinap, 2015-03-26:
@thenickdude : #177 wont help here without quite large LEDs code rewrite. LEDs need whole timer (in 8bit mode). But it may by possible to move things around so that function that don't need timer are on remaining LEDs timer pins.

hydra, 2015-05-15:
See #375 and #884

closed 2017-02-25 by hydra