cleanflight/cleanflight #470: Verify Spektrum-Sat Binding on Flip32+

pulsar256, 2015-02-03:
Unverified steps to Bind:
```
feature -RX_PARALLEL_PWM
feature RX_SERIAL
set serial_port_2_scenario = 3
set serialrx_provider=0
map TAER1234
set spektrum_sat_bind = 3
save
# Saving
# Rebootin
```

after Power-Cycling the Sat should be blinking and waiting for a TX to pair with.

pulsar256, 2015-02-03:
bind attempts were unsuccessful when using the dedicated Spektrum-Sat header. Seems it is not routed to USART2 but USART1 which conflicts with MSP over USB.

stronnag, 2015-02-03:
On the flip32, the sat port is known to be incorrectly wired and you have to use pin #4. OTOH, I've never got sat binding to work using pin 4 either.

Zappes, 2015-02-03:
I just connected the Orange R100 as follows:

- orange vcc -> flip32+/boot/pin1 (3.3v)
- orange gnd -> flip32+/rc4/gnd
- orange txd -> flip32+/rc4/signal

Then I used spektrum_sat_bind=3 and the Orange R100 bound to my DX6i without a problem.

closed 2015-02-03 by pulsar256

fiveangle, 2015-02-16:
FYI: the advice of stronnag above only works for rev 5 of Flip32. Latest Rev 6 (yes, all of them, as it seems every clone has replicated Paul's jacked up Rev 6 design) has SAT signal connected to MPU_INT, so it's even worse than Rev 5 and requires a more difficult trace cut to disconnect MPU_INT and new surface trace to rc4 to function. 

-=dave