cleanflight/cleanflight #476: Airplane RTH support

universam1, 2015-02-04:
Next step for CF would be to support fixed wings GPS navigation for RTH and Pos hold like Baseflight does already.
http://fotoflygarn.blogspot.de/2015/01/baseflight-fixedwing.html

Please bring CF to the next level!
https://github.com/multiwii/baseflight/blob/master/src/fw_nav.c

theailer, 2015-02-04:
And custom servo mixer if it isn't already in.

universam1, 2015-02-04:
+1 for airplane custom mixer. This is the major requirement for serious airplane application.

Akfreak, 2015-02-09:
I would love to see better plane support for Cleanflight.

nppc, 2015-02-09:
Yes, GPS RTH is very useful feature for the Gliders.
Custom mixing for "Full House" glider is very handy. :)
Thank you!

hydra, 2015-02-26:
Blocking, GPS PH needs improving first.

hydra, 2015-06-24:
I removed the blocked label.  However anyone working on this should be aware that @digitalentity is working on new navigation code - if you work on this please work together so that you don't duplicate any effort.

digitalentity, 2015-06-24:
Thanks @hydra! My code is focused on multirotors for now, but I have fixed wing in plans and wrote my code keeping that in mind.
If somebody is willing to help - you are most welcome! Just go to https://github.com/digitalentity/cleanflight and look for FIXME's in code - they are mostly marking the spots where fixed wing navigation code should be.

sppnk, 2015-08-21:
@digitalentity and @hydra ,

 Will we have RTH for airplanes without baro and compass in next versions?

 I have some CC3D that would be great with that feature and baro and mag are not mandatory for airplane RTH functions. It have been done in Multiwii 8 bits by PatrikE and I think in BaseFlight too (but baseflight cant be installed in CC3D boards).
Thx

digitalentity, 2015-08-21:
@sppnk 
I have fixed wing support in mind, but I want to develop full support for multirotors for now. Thus I won't expect airplane support any time soon, not until multirotor code reaches stable state.

sppnk, 2015-08-21:
Ok, thx anyway. We will wait. :-)

digitalentity, 2015-08-21:
@sppnk waiting won't make things go faster, helping will :smirk:

sppnk, 2015-08-21:
Oh, please dont missunderstand me. I am not very good in programming skills, but I will test your code and report any issues or success as soon as you code it . For now it is the way I can be more usefull.  

And later I will try to help when I could understand how this github thing works (today is my first day here) :-)

digitalentity, 2015-08-21:
Oh, I see. Are you able to test multirotor code or only fixed wing? Sooner I get multirotor code working- sooner I'll get to developing airplane code.

sppnk, 2015-08-21:
For now I am just testing two airplanes, one with PatrikE Multiwii RTH version and other with cleanflight. I have also a multicopter, but it has a Multiwii 8 bits board. 

I am thinking of build another quadcopter to test cleanflight and of course your navigation code, as I have another CC3D board and enought hardware to build it. Just give me two or three weeks.

closed 2017-02-25 by hydra