cleanflight/cleanflight #483: Verify compatibility with Paris AirHero32.

hydra, 2015-02-06:
http://www.multiwiicopter.com/products/paris-air-hero-32

hydra, 2015-02-06:
Hardware required, please donate!

adamhope, 2015-02-17:
I've got one and am happy to try it out but I'd like to get an idea of the risks first ;-)

hydra, 2015-02-20:
hardware received

justadude-OG, 2015-02-24:
I am watching this because I just bought one of those Paris air hero32 's and wanted to run it on clean flight. Keep us posted.

justadude-OG, 2015-03-13:
Any Progress here Hydra?

bobrc11, 2015-03-20:
I have one as well, that I'd really like to be able to use.

ShikOfTheRa, 2015-03-25:
Yep - another+1 here!
I really like how this one performs under SPI on BF. 
Would really like to see this one with CF too

If you need hardware to dev/test I can my spare over

hydra, 2015-04-08:
Support added in d3d9721e9195f40752d7aa2cdefe6923fb1c5426

Please test and report on here what works and what does not.

hydra, 2015-04-08:
CI builds now available:

http://cleanflight.memoryleaks.org/builds/2015-04-08/861_d3d9721_cleanflight_NAZE.hex

http://cleanflight.memoryleaks.org/builds/2015-04-08/

AlouetteIII, 2015-04-16:
No success with Sbus X4R from RC pin 4?  I am not able to get it to receive Sbus - If i select Rx Serial & SBus - save ; and go to Receiver panel, all 1500 (Rx is bound)  - if i go back to Config the Sbus setting has reverted to multiwire - nothing selected under Receiver Mode (self un-checked) ; this is occurring during the Save & Reboot - the Sbus is self-deselecting - only for Sbus - any other Rx mode is sticky.

This board has a built in SBus hardware inverter on RC pin 4

AlouetteIII, 2015-04-16:
When Arming by Yaw stick Right, Min Throttle 1 sec - UI "modes" does not show armed (green panel) - ideally this ARMED status flag would show on ALL tabs (top bar of UI says "ARMED" ) - for noob safety when learning to arm and dis-arm as first controller (props off).

AlouetteIII, 2015-04-16:
manual motor drivers/sliders bug - if running motors at say 50% - and move/click across to Receiver tab to check RC - then reduce Throttle in Rx Tab and disarm - when return to the motors tab the manual motor control sliders are all up in the position where the user exited the Motors tab (in this case at 50%) - having never moved the sliders off min  (OSX 10.10.2 Chrome Version 41.0.2272.118 (64-bit)

hydra, 2015-04-16:
Can you keep this thread for issues that are specific to this hardware.  Also please use the configurator issue tracker for GUI issues or feature requests.

I think there is already an issue raised for the motor tab bug.  If not please raise it in the GUI issue tracker.

Yes, showing system status indicators in the GUI would be great.

AlouetteIII, 2015-04-16:
Noted !

C-PPM tested and working with D4R-II using RC Pin1 single wire

AlouetteIII, 2015-05-21:
Flexport not active for I2C, specifically  GPS/MAG for Return to home functionality

hydra, 2015-05-22:
I2C and USART3 support added to the flex port.
[image]

AlouetteIII, 2015-05-25:
I2C oLED working in endless loop page scroll mode - 1.9.0 - yes
I2C LCD_CONF feature not working as used in MultiWii - not sure if this has been ported over from MultiWii -  (PID tuning by LCD & sticks - this is the main reason for using a oLED as it means PID tuning can be done in the field with no phone or PC)

Paris Sirius I2C MAG working/detected but not in correct XYZ - should be 
  #define MAG_ORIENTATION(X, Y, Z)  {imu.magADC[ROLL]  =  -X; imu.magADC[PITCH]  =  Y; imu.magADC[YAW]  = Z;} 

Paris sirius I2C GPS not working as per #define I2C_GPS (not sure if I2C_GPS has been ported over from MultiWii 2.3 yet?)

cjbeck, 2015-05-27:
So, SBus doesn't work, correct?

AlouetteIII, 2015-05-27:
Sbus is working provided you first select       SERIAL Rx      on UART2 in Ports tab

closed 2017-02-25 by hydra