cleanflight/cleanflight #487: s.bus support regression futaba 3008

knowmercy, 2015-02-09:
Back in version 1.0.0 my futaba 3008 using s.bus with a signal inverter was working well. It is no longer working at all. As best I can tell it happened in this commit: https://github.com/cleanflight/cleanflight/commit/5401092afaabf
3b19391c3da8d3fb3fc211df889#diff-cee35236599ccaf4e7c09403d9b593e9

I'm using a naze and a flip32 or flip32+, the behavior is always the same.

tricopterY, 2015-02-09:
Is feature 'INVERTER' enabled? The target has to be  AFroNaze.

knowmercy, 2015-02-09:
No, it is not enabled. I just followed the documentation on the site. I'm using a hardware signal inverter.

tricopterY, 2015-02-09:
Sorry, my mistake - there is no feature as INVERTER, or I do not see it. Unplug your hardware inverter and try again. There is code that inverts signals of a particular serial port, if AfroNaze is the target type. Not sure about Flip type, but if it's clone of CC3D should work.

knowmercy, 2015-02-09:
That's my point... this used to work. Back in november I flashed the cleanflight firmware on my quad and all was well. I went to fly today and cleanflight told me I couldn't use the old outdated firmware so I upgraded and now it doesn't work.

tricopterY, 2015-02-09:
I am struggling with same concepts - I am just poking around the code to see the application of software Inverter for particular target type(s). They discourage use of 'Issues' for support. Best is to use IRC channel.

hydra, 2015-02-16:
Ok, i did a bit of work on this today, try compiling a new version from the master branch.
A different approach to detecting timeout has been implemented.

Here's the timings taken from my scope for an X8R.

[image]

[image]

Roeland54, 2015-02-17:
Compiled a .hex from the master branch. The problem I have stays the same. SBUS works except receiver failsafe. The channels just lock into position. Tested low and high speed mode on a Futaba R6308SBT. If there is something else I can try, let me know.

hydra, 2015-02-20:
in the configurator the channels will stay at their last values unless the failsafe code overwrites them, this doesn't mean that failsafe is not detected.

enable failsafe feature, arm with props off, view motor tab, put throttle at mid, observe motor values are being updated as you move the board, turn off TX, observe motor outputs change and then motors are disabled.

during failsafe (when armed, and afterwards) you should also be able to see on the received that that the first 4 channels are updated.

please confirm.

hydra, 2015-02-20:
btw, I verified and dumped the failsafe and signal loss bits used on futaba are the same as the frsky gear using a logic analyzer.  it's possible but unlikely that the 3008 is different but the R6308SBT definitely worked when i tested it.

hydra, 2015-02-20:
[image]
[image]

Those are screenshots of the R6208SB's.  The digital channels were on (bits 0 and 1) and then the tx was turned off (signal loss and failsafe bits 2 and 3).

Roeland54, 2015-02-21:
Ok did some testing. I can confirm the cleanflight failsafe function is working as it should work. 
cleanflight is detecting the failsafe state of the sbus receiver and is reacting like it should. 

But in my opinion it should be possible to use the failsafe of the receiver when "feature failsafe" is not enabled in cleanflight. At this point al channels lock into position instead of reading the sbus channel state. Which is at this point the configured failsafe state of the tx/rx combo. 

If this is the expected behaviour we should add a warning to the documentation. Because this is a significant difference between Sbus and PWM - rx handling.

Thank you hydra for putting so much time into the cleanflight project I really like it!

hydra, 2015-03-03:
A new branch has been created with a potential fix.

https://github.com/cleanflight/cleanflight/compare/sbus-rx-failsafe

Please compile and test with your Futaba RX and let me know what you see on the receiver tab.

hydra, 2015-03-04:
Merged to master.

I verified that when the RX (in my case, FrSky X4RSB) reports failsafe mode (via the flags in the data packet) that:
a) failsafe is correctly triggered when enabled in the FC
b) that the channels update to show the channel values (in the case of the X8R the value is 880).

Closing this issue.

closed 2015-03-04 by hydra