cleanflight/cleanflight #488: Change default of Failsafe

andymac4182, 2015-02-09:
Few people in our group have forgotten to set failsafe recently after setting up a new quad or upgrading firmware. It would be great to have a default of it turned on and set to 1100 as the failsafe throttle. This way at least if you forget to set it there is something there to ensure your quad doesn't just fly away and never come back.

fiveangle, 2015-02-09:
Problem there is on the rare chance it gets armed on the bench and they shut the TX off (not realizing FS is enabled), wouldn't the motors spool up ? I think it's better to lose the quad than to lose an eyeball :)

There's some other FS issues open and a lot of improvement to make in this area, but I think it will need some open dialog will the devs to sort out which is the best direction to go. 

-=dave

andymac4182, 2015-02-09:
As far as I am aware failsafe is only used once armed.

snikch, 2015-02-09:
Just thought I'd chime in here and say that this exact scenario occurred to me the other day with my new tri. I stupidly forgot to disarm and turned my TX off while walking the tri inside. It spun up to 1500 on throttle and opened up a few lacerations on my fingers. I immediately turned fail safe on! It seems like a sane default to have.

MJ666, 2015-02-09:
I think if the copter is armed but motors are off the failsafe should not cause to spin up motors again since the copter is likely siting on ground already. Motor should only keep spinning if they are up and running.

FinalGlide, 2015-02-10:
Just set failsafe to 1000 by default then.  Sorry I added this same issue just before :(

fiveangle, 2015-02-10:
MJ666: just because motors are not spinning, doesn't meant it's not armed and it certainly doesn't mean it is sitting on the ground. The code clearly just looks at failsafe_timer.  It doesn't do anything fancy (which would probably be nice) like see if accelerometer data is pegged at 9.8ms^2 along a single vector ;)

FinalGlide: Yeah, that would work.  Changing the default for it to fall out of the sky is better than having it fly away I suppose.

-=dave

MJ666, 2015-02-10:
@fiveangle 
In my view if the copter is still armed but motors are already off they should not start spinning again if there is a failsafe situation.

cjbraden, 2015-02-10:
I agree with @MJ666 point just now that if the motors are already off, they
should not start spinning again. Another common failsafe condition is when
the machine has already augered into the ground and antennas are no longer
in a good place reception-wise. I had mistakenly thought through my
failsafe options that they would likely occur in the air, and that I want
the machine to somewhat gracefully fall to the ground so I engage
autolevel, and throttle to 25%ish. In practice though, when I'm about to
crash I cut throttle to zero, hit the ground, and all of a sudden my
fialsafe engages on the ground and I burn an ESC trying to spin back up to
25%.

strepto42, 2015-02-10:
@cjbraden I think the other problem is that the default is to engage failsafe for 20 seconds, which is more than enough to fry things, and also long enough to cause a partial flyaway (for want of a better term).

Perhaps defaults should be minimal motor speed (enough to autolevel the quad to minimise damage, but not stop it descending) and a very short duration like 1-2 seconds.

fiveangle, 2015-02-10:
So if I throttle-low entirely, the suggestion is for the craft to disarm ?  Am I missing something ?  This means that anyone doing acrobatic flight will have their craft fall from the sky when they throttle back up.

Cleanflight is not designed to be solely a camera platform flight controller FW.

-=dave

fiveangle, 2015-02-10:
@cjbraden: I have no issue with setting failsafe to "throttle off" by default.  I have a huge problem with the propsal that recovering from failsafe is not possible if throttle is "off" when failsafe engages, because the FC will simply disarm.

-=dave

cjbraden, 2015-02-10:
@fiveangle I agree with you, and I don't think there is a one-size-fits-all solution. I think the default should be "throttle off" and everything else as is. I was just relaying a real-life scenario that happened to me. "This means that anyone doing acrobatic flight will have their craft fall from the sky when they throttle back up" - this would only be the case IF failsafe triggers while it's throttled down, it the scenario I mentioned earlier. Not just every machine that throttles to zero falling out of the sky... Again, after giving it more thought, I don't think it's a good solution, I just wanted to add to the discussion, that if you use anything other than "throttle off" as your failsafe. be aware that machines on the ground far away from you have a tendency to lose radio signal.

fiveangle, 2015-02-10:
"this would only be the case IF failsafe triggers while it's throttled down".

Correct.  If you are flying around in a 250 miniquad through the trees, and you happen to have throttle at PWM 1000 when you loose RX signal, your craft disarms and never recovers from failsafe.  I'm struggling to see how this is a good idea.  No, the failsafe should run its course (however it is configured) and either allow recovery (if RX signal recovers), or disarm once the failsafe_timer has expired.

-=dave

fiveangle, 2015-02-11:
To summarize the parts of the discussion so far that I think most agree on:

Current behavior has FS not enabled by default, resulting in craft fly-aways for new users who do not set it up.

Proposed change: enable FS by default but set default failsafe_throttle to be throttle-off ('failsafe_throttle <= 1000') instead of '1200' in config.c:
'''c
    currentProfile->failsafeConfig.failsafe_throttle = 1200;         // decent default which should always be below hover throttle for people.
'''
Effect:
1. prevents flyaways for new users (instead, killing motors falling out of the sky)
2. requires those that want to enable graceful FS to test their craft for correct decent failsafe_throttle value to use

I see this as a win-win; it will prevent flyaways for new users, and will force those that want graceful FS to test and correctly configure FS to work optimally for their craft.  The current behavior promotes newbie flyaways, as well as sub-optimally configured FS for those that just enable it without testing.

-=dave

fiveangle, 2015-02-11:
One problem that arises, is the fact that FS kicking in won't auto-level the craft (motors will be off), so if singal recovers before FS timer expires, the craft will likely be unrecoverable for any users flying in rate mode.  Those flying in Angle or Horizon it won't be affected much at all, since the craft will auto-level on signal recovery.   However, those flying in Rate mode without having done any FS configuration will have their craft throttle-off when signal is lost, rather than use the last received signal values.  This will need some thinking of the tradeoffs.  I don't see a scenario where we aren't trading one demon for another. The question is, which is the lesser of the evils ?

-=dave

mosh1, 2015-02-28:
The default of the FC holding last command on FC failsafe caught me by surprise when I ran Spektrum Satellite (non full RX without its own failsafe) and lost signal (this was on Baseflight). Most full commercial receivers if not programmed, have the default to drop throttle and hold all other commands on signal lost. Holding throttle on signal lost is something that no full receiver has ever done. Having Cleanflight drop throttle to 1000 on no PPM signal will mimic the behavior users have been accustomed to with full receivers when they only have a satellite attached and is also a saner default (safe and no fly aways).

Folks running full receivers won't ever hit the FC failsafe situation unless either they program the receiver failsafe to also trigger FC failsafe - or have a receiver wire pull from the FC so the FC doesn't see PPM signal. In that case it still seems reasonable that the user set up any signal lost failsafe by the RX (angle mode, no or partial throttle) - and have the FC failsafe (if hit due to pulled wire) to throttle down on default.

fiveangle, 2015-02-28:
Agree 100% @mosh1

I think the idea of adding disarm automatically on signal loss to that sequence should be via a new issue if someone else wants to raise it (and reference this one).

-=dave

hydra, 2015-03-05:
Ok, I think enabling FS by default and using low throttle is generally safer than a fly-away.   If we want to handle special cases please create new issues and discuss there.

@mosh1 Please update the PR to include an update to the failsafe documentation so we can merge for 1.8.x.

mosh1, 2015-03-06:
Done.

hydra, 2015-05-15:
Failsafe is enabled by default in master branch now.

If anything else needs to happen please open a new issue/PR and reference this.

closed 2015-05-15 by hydra

hydra, 2015-05-19:
All involved, please see #912

hydra, 2015-05-19:
Please see cc5c736362793aaa124442275a7b7b6078d75433

ctzsnooze, 2015-06-17:
I've just been flying with a flakey old Lemon Sat that several times went failsafe in the FC.  It was a long walk to the craft.  I was sure I had radio reception but could not take off. 

So I was wondering how to disable this feature:  "After the failsafe has forced a landing, the flight controller cannot be armed and has to be reset."  

I would like the ability to make it automatically disarm on landing but permit it to be re-armed, rather than locking up uselessly.

The reasons include:

- sometimes it falls into a tree, and sometimes we can get it out by intermittently spinning motors. At the moment I cannot spin them to help get it out because I cannot re-arm

- sometimes I can walk close enough for signal to re-establish, and just want to lift off and fly home, rather than walk a lot further to reboot.

- sometimes it ends up on flat roof or inaccessible place that I could fly away from but not climb up to.

Personally I think this should be configurable, somehow, and not be too hard to do.

strepto42, 2015-06-17:
I didn't realise this was a 'feature' now, but in hindsight it does make sense because I've seen this behaviour too.

I definitely think it should be configurable. Transmitters can move. Stranded quads that have failsafed into trouble cannot.

Shadow-MasterTf, 2015-09-21:
I totally agree with ctzsnooze what is the point of the Fc locking up?

ctzsnooze, 2015-09-22:
The latest master version, with all the new improved RxFail code, now can be re-armed 30s after a failsafe landing.