cleanflight/cleanflight #489: Rates Overhaul

dave-pitman, 2015-02-10:
First, a little background.  I'm not an expert at this, so I may be wrong. Please correct me if I am so that we can get to the heart of the matter.

What we have now.

RC Rates and Expo   and   PID Rates

According to the multiwii wiki (http://www.multiwii.com/wiki/index.php?title=RC_rates_%26_expo)  RC Rates controls FC sensitivity and reaction to pilot input across the entire flight envelope.

RC Expo, if used, creates a 'flat' spot of varying degree in the center of the control line so as to soften sensitivity when  the pilot is not intending to add input in any direction.

PID Rates are supposed to allow full deflection at axis extremes are to allow full and fast rotation for flips and rolls.

Putting it together.

Currently, the RC Rates and PID Rates, while intending to target different parts of the flight envelope, at some point overlap. What we want is to control each axis. Each pilot will prefer a different response to stick input. The goal should be to make this adjustment to how the aircraft reacts to stick input as simple as possible.  Adjusting RC Rates and Expo, then adjusting PID Rates, and possibly back to RC rates, etc., is not super efficient.

What is proposed is to have a Rate, Expo, and Curve for each axis only.  
Each axis' sensitivity to input, soft (or null) zone in the center, and curve shape when combined would be all that should be needed without the current back and forth.

A novice or even AP pilot will probably want less sensitivity, a larger null zone, and a linear curve profile.  While an acro pilot would obviously want more sensitivity, less or no null zone, and perhaps a steep curve at the max endpoints.  Both of these extremes should be easier without the current back and forth.

Again, I'm going by the documentation.  Perhaps the calculus in the code tells a different story.

nebbian, 2015-02-10:
I'd be happy with just:
Roll Rate
Roll Expo
Pitch Rate
Pitch Expo
Yaw Rate
Yaw Expo
Throttle Expo

This would allow full tuning of each axis independently.  I don't see a need for curves for each axis.

dave-pitman, 2015-02-10:
It seems that PID rates were added at some point to allow greater response at the extreme endpoint for aerobatics.  While expo has an influence on the control path,  I'm not sure that without the ability to shape the curve at the extreme endpoint that the acro guys would be happy.   The default linear curve would be the same as no curve and could probably be left untouched for all but aggressive flight.  But again, this is all theory on my part.

tracernz, 2015-02-10:
>RC Expo, if used, creates a 'flat' spot of varying degree in the center of the control line so as to soften sensitivity when the pilot is not intending to add input in any direction.

That's not really correct. The idea of expo is so that you can have very high rates near your endpoints while still maintaining fine control around centre. What you have described above is solved by deadband.

dave-pitman, 2015-02-10:
>The idea of expo is so that you can have very high rates near your endpoints while still maintaining fine control around centre. What you have described above is solved by deadband.<

Very good.  Do you think that Rate and Expo alone would be sufficient to achieve high enough response at endpoint while not being too sharp in the center without a user plottable curve?

nebbian, 2015-02-10:
Just a rate and expo would be enough for 95% of users I'd imagine, and for the remaining few they could always put a curve into their radio.  I don't see four extra curves as being worth the added complexity.  The nice thing about expo is that it's smooth: the rate increases in a smooth manner as the stick is deflected which is necessary for complete control.  I think that the extra Pitch/Roll rate might have been an attempt to split the rates for the different axes (so that flips/rolls were quicker than full stick yaw), which could have been solved a different way.

dave-pitman, 2015-02-10:
Probably just Pitch and Roll curves, if any.  I don't think that yaw or throttle would need it for sure.

tracernz, 2015-02-10:
>Very good. Do you think that Rate and Expo alone would be sufficient to achieve high enough response at endpoint while not being too sharp in the center without a user plottable curve?

Yes, you only need look at collective pitch helis. They have much higher roll rates and sharper response, yet need much finer control around centre. Everybody manages just fine with only rate and expo controls.

dave-pitman, 2015-02-10:
Good example, thank you.

Ok.  How about from the coding side.  Getting rid of RC Rate and RC Expo,  and setting up Rate and Expo per axis only.

Doable?

vadav, 2015-02-10:
dave, yaw would benefit of both rate and expo..   so im with nebbian on this:

I'd be happy with just:
Roll Rate
Roll Expo
Pitch Rate
Pitch Expo
Yaw Rate
Yaw Expo
Throttle Expo

dave-pitman, 2015-02-10:
@vadav   Yes, all axis would have controls.  Yaw was only left out of the 'curves' discussion, it would definitely need rate and expo.
Throttle expo would be unchanged.

vadav, 2015-02-10:
@dave-pitman Sounds good, so all we need now is a codeYedi?   =D

hydra, 2015-05-15:
Yaw Expo has been added in master branch (1.9.x).

Can someone summarize this and say what is still needed?

hydra, 2015-05-15:
Please see https://github.com/cleanflight/cleanflight/issues/210#issuecomment-102412831

closed 2015-05-15 by hydra