cleanflight/cleanflight #490: Openpilot V9 GPS support

andymac4182, 2015-02-10:
I believe it uses serial to talk. It would be great to add this support as it includes the magnetometer as part of the serial data reducing the requirement for a second cable and i2c.

hydra, 2015-05-15:
We're gonna need more details for this even to get looked at, including:
1) links to products
2) datasheets, product manual links.
3) programming manuals for the protocol.

That said, no-one else seems to want this and there's already good alternatives that are fully supported and open source; adding a proprietary GPS protocol doesn't make sense for cleanflight codebase or it's ideals.

If it's not proprietary then please give further details about the protocol and we can re-examine this decision.

I think I'll close this but people can comment on it if they really do want support for it.

closed 2015-05-15 by hydra

Trussroads, 2015-05-17:
If people are asking for this based on wanting general performance gains, then please look at "Navspark" as the GPS to add alternatively. 
This is a centimetre accurate receiver such as the Swift Navigation "Piksi" and I have a few on my bench for when I have time to investigate integration myself. (The quality & performance is excellent)
This would be a good choice for Cleanflight in my view and the price advantage is a big positive, as is that it supports Beidou, and not least of all that it is an open project.

All the original project overview info from its Indi launch here https://www.indiegogo.com/projects/navspark-arduino-compatible-with-gps-gnss-receiver
Skytraq specific chip info here http://www.skytraq.com.tw
Can be purchased in many forms in the webstore here http://navspark.mybigcommerce.com

Please let me know if I should have posted these comments elsewhere? Apologise in advance if so.. 
I also am not familiar with Github so would need guidance to create a PR if that is what you would like done with this info?
Thanks all.