cleanflight/cleanflight #491: Sonar/barometer conflict

jistforfun, 2015-02-10:
While I want to say thank you for adding sonar to the sensors tab, this helped me to trouble shoot a bad sonar module, the sonar is over riding the baro. 

When sonar is connected via pins 7/8 on full naze32, soft serial 2 the baro graph follows the sonar graph. The sonar reads and responds correctly in the sensors tab, but if the quad is move horizonally over a ledge with out a change of altitude the baro graph will plunge following the sonar graph.

When ever the sonar module is plugged in altitude hold will not function regardless of weither or not sonar or baro mode are selected.

If the sonar module is unplugged baro hold will work with either baro or sonar mode selected.

It would appear that both the sonar and barometer are sending data to the same point conflicting with each other.

thenickdude, 2015-02-10:
Looks like the "baro" graph is actually an estimated altitude graph. Cleanflight automatically picks the source it thinks will be the most accurate when it estimates the altitude, which means that the sonar will be used to measure short distances and the baro for longer ones. I think the handover point is at the 2m range.

enly1, 2015-02-10:
Yep, 0-2m the sonar takes precedence over the baro. (assuming its not
tilted > 25 Degrees and returning 0)
Not sure from memory when I last looked at the sonar code if it then
utilises Baro reading when its a 0, or just skips the sample / alt
estimation changes etc.

rosek86, 2015-02-10:
I think it'd be better to show raw baro data in on the sensors tab.

The current implementation is confusing, the label says 'Barometer' but the
graph shows combination of baro & sonar. Possibly there should be separate
MSP command to read raw baro which can be use in the configurator?

jistforfun, 2015-02-10:
At the moment Sonar is always on, from the moment it is plugged in, though not function as a sonar alt hold selected or not.

I would like to see sonar and baro read seperately on the sensor tab, raw data.

Then allow  baro mode to be selected independently of sonar and sonar mode independent of baro mode, unless both sonar and baro are modes are both selected at the same time in the modes tab.

This would allow for three altitude hold modes:

Baro
Sonar 
Baro + sonar

Lastly it would be helpful to PID parameter that would adjust how quickly the multirotor reacts to a change in sonar, just as the barometer does.

Thank you for all your hard work!

hydra, 2015-02-10:
I have previously thought the same with regards to the separation of baro/sonar.

Also agree that the pids for the sensors might need to be different because sonar and baro sensors updates happen at different frequencies and sonar readings are more accurate.

This is also true for LIDAR.

jistforfun, 2015-02-11:
Lidar! Is there a possibility lidar support will be added to CF! Maybe optical flow next?

Long term I am wanting a FC that will support multiple sonar sensors for obstacle avoidance an allow a multi to operate in tight spaces. Imagine being able to use a small multi to do internal HVAC inspections or other unpleasant jobs in confined spaces.

I realized this is beyond the scope of what is possible for a naze32, but totally doable for FC's soon to come.

In any case sounds like the Sonar/ baro conflict is understood.

Thank you guys for your hard work!

klHaribo, 2015-02-12:
When I aktivate modes sonar and baro at the same time, the motors stops immediately (in flight).
When i pull throttle to zero, motors starts to rotate at idle speed.
(tested with different versions)

jistforfun, 2015-02-13:
Maybe by 1.73? 
Hope hope...

jistforfun, 2015-03-18:
I see Hydra modified this issue a few days ago, does that mean it is being worked on?!

I feel bad to keep pushing this issue will all the other great improvements in 1.8,
but Sonar is still a feature I would like to use when it is working again.

Thanks again for your great work.

hydra, 2015-05-12:
If it was being worked on then it would have in-progress label assigned to it.

closed 2017-02-25 by hydra