cleanflight/cleanflight #492: Greater delay before communicating with OLED display

SlyMike, 2015-02-10:
in reference to closed issue #462 

I am on my second OLED display, the first one went in the bin!

The model is Crius v1.2. it has the R/C timer on the reset pin and correct signals broken out to the PCB.

The first unit worked well for a few days and then became flaky in operation, the second did not work at all. After removing the cap from the R/C timer on the reset line I managed to get it to show some data (and lots of garbage) once in every 10 or so power cycles, mostly it just locks up the i2c bus and the FC fails to start.

Also tried this display on v1.50, 1.60 and 1.7.1 with the same outcome.

I'm just thinking that maybe some OLED displays need more time to start up than others and giving a second or two after initially plugging in the battery may help the 5v line to rise up and the OLED to come out of reset properly.

snoekieboe, 2015-02-10:
I had the same stuff happen with my banggood 0.96 inch I2C display. Sometimes the FC just won't boot properly with the screen attached whilst other days it works like a charm. Never found out why it did this but since i swapped to a new Naze board the problem has never occured again.

SlyMike, 2015-02-10:
Just tried powering the OLED from a separate PSU tied to FC ground and the problem persists.
So, in my case it's not down to dodgy wiring or uBEC.

@hydra, if you want me to send you the OLED in the post, you can have it for the greater good of the project.

hydra, 2015-02-17:
Yes that would be very useful. Please drop me an email and I'll give you my address.

SlyMike, 2015-02-18:
@hydra Message sent!

SlyMike, 2015-02-19:
@hydra Sent n emailed!

hydra, 2015-02-20:
So i received your screen, i plugged it into my new fc and it worked fine, no modifications or anything.  I just unplugged my old hacked up crius and plugged in the one you sent @SlyMike and it works perfectly.

The one you sent me was different to my one, yours had 6 smt components on the right hand side where mine only had 4.  Mine was on a blue PCB, yours was on a black one.

hydra, 2015-02-20:
I have seen the issue with regards to the bootup problem, i agree that we should continue to detect the display after power up and only use it if it's present.

SlyMike, 2015-02-20:
Ummm. Glad it worked for you but bemused that it didn't work for me!

The first one I had worked great for a week and then declined over a few days to the point that I threw it away. I ran them both at 5 volts from the FC board. The one I sent you never worked for me at all, however the hobbyking one worked fine after this mod...

(http://www.hobbyking.co.uk/hobbyking/store/__50096__Multiwii_MINI_OLED_Display_Module_Dual_I2C_128x64_Dot_MWC_MINI_.html) worked fine after this mod...
[image]

IDK, Maybe all Naze32 boards aren't made equally?

Enjoy your new screen and close this issue when you feel appropriate.

Cheers, Mike.

hydra, 2015-02-20:
Can I use that image in the documentation?

SlyMike, 2015-02-20:
Yes you can.

hydra, 2015-02-20:
great thanks!

SlyMike, 2015-02-20:
:-)

hydra, 2015-04-18:
Fixed with ebd5475a55dcb0954d9d97c0a10b14b0bd2d6c43

closed 2015-04-18 by hydra

alvins82, 2015-07-03:
I have replicated the issue #462 and #392. The issue occurs if you have OLED and try to power naze through USB only. If I hook up battery first (i.e. so OLED boots up) and then connect USB then there are no issues.