cleanflight/cleanflight #493: Failsafe default setting

FinalGlide, 2015-02-10:
By default the Failsafe is not enabled.  This to me is silly and dangerous for those who are new to the naze and not aware of the failsafe feature.  If you are using a Spectrum Sat receiver and have a lock out then the quad will hold last setting.  If you are using a full receiver and a receiver wire comes loose (this happened to friend last weekend) then the quad will hold it's last position.  This is almost always going to end up with a fly away.  I recommend that failsafe is enabled by default.  This then brings the issue that the motors will now spin when failsafe is triggered.  If a new pilot was to turn off their transmitter then the motors will spin up and possibly hurt someone.  I recommend that failsafe cut the motors by default.  That way new pilots won't have a fly away and no one will get hurt when they accidentally when they turn the tx off first.  Once a person learns how to adjust the failsafe they can then change it to suit their tastes but until that point there is no chance that they will have a fly away or hurt themselves.  Considering we will soon be having lots of Flitetest fans joining the Cleanflight team soon I think this needs to be addressed.

andymac4182, 2015-02-10:
Similar to https://github.com/cleanflight/cleanflight/issues/488

hydra, 2015-02-10:
yeah, let's close this issue, further discussion should take place in #488 

thanks for feedback @FinalGlide

closed 2015-02-10 by hydra