cleanflight/cleanflight #494: CC3D internal Sbus inverter does not work

Arcicorsa, 2015-02-10:
If select Sbus receiver on UART1 not activated integrated signal converter and must be used external converter. Thanks for repair.

hydra, 2015-02-10:
yup, it has not been implemented yet.  some code exists but it's not fully tested and requires more work.

JBFUK, 2015-02-10:
I wouldn't say this is an enhancement, it's pretty critical if the CC3D board is to be supported.  Very disappointed that I can't use cleanflight because of this.  Looks like I will have to flash back to openpilot for now :(

hydra, 2015-02-11:
To clarify: Development is iterative. If it was done and there was a problem it would be a bug. If its never been done it's a feature/enhancement.

It'll get done, things require time.

JBFUK, 2015-02-11:
Understood. I'm not a coder (although I can find my way around scripting
and things like arduino code), if there's anything I can do to help with
this, testing etc let me know as I'm very eager to get away from
openpilot.  I had a look through the serial code last night, I saw the
'inverted' argument in some of the functions but have to admit I was a bit
lost.  Should have brought a naze board...

Arcicorsa, 2015-02-11:
To Hydra: I make simple change in inverter.c . Change is on line 36. Here is correct code:

.cfg = { INVERTER_PIN, Mode_Out_PP, Speed_2MHz } 

Now is converter working :-) .

hydra, 2015-02-11:
Excellent,when I looked at it a couple of weeks ago I didn't think there was much else that needed doing. Thanks for the code and confirmation.  I will merge and test this soon.

hydra, 2015-02-11:
Documentation will need updating too.  Any chance of a pull request @arcicorsa ?

Arcicorsa, 2015-02-11:
So far I've never done it but I will try to pull request on it in the near future. Also, my English is bad for documentation update.

hydra, 2015-02-11:
Tested here with FrSky X8R on main port, no external hardware inverter needed

closed 2015-02-11 by hydra

JBFUK, 2015-02-11:
Not wanting to sound stupid but as a newbie I have to ask this once - how do I compile and test/use this prior to a new release being available through the website?

hydra, 2015-02-11:
See docs.

hydra, 2015-02-12:
Release candidate for v1.7.2 is available with this fix.

https://github.com/cleanflight/cleanflight/releases/tag/v1.7.2

@JBFUK please test and let me know if it's ok for you.

JBFUK, 2015-02-12:
I pulled the latest version via git after you logged off IRC last night/this morning; compiled using CC3D target and OPBL flag, flashed via GCS and it works.  I presume this will be the equivalent to using the pre-compiled 1.7.2 release referenced above?  If not I can re-flash with 1.7.2 tonight and confirm everything still works.

FYI I'm using an OrangeRX 3 channel + SBUS receiver. Cleanflight configurator shows all channels and they appear to be working.

Thanks for the quick turn-around.  Happy that I can start using cleanflight now - I like the development model you are running with; very effective and open.

Davhed, 2015-03-05:
I apologize for putting a question on a closed topic. I was unable to find a definitive answer to this on the MultiWii forum or through a Google search. Does this commit mean that SBUS inverters are no longer needed for any boards, or just CC3D? I am curious if I will need an inverter for SBUS FrSky to Naze32

JBFUK, 2015-03-05:
I believe it's only the CC3D that has a built in inverter but someone else
may correct me on this.

Davhed, 2015-03-05:
That makes sense, thank you for the clarification. Naze32 SBUS inversion would be a sweet enhancement then. Awesome work on CF!

hydra, 2015-03-05:
Boards with F1 processors need external inverters. CC3D has one.
Boards with F3 processors do NOT need inverters. The CPU itself can do it.

Davhed, 2015-03-05:
OK thanks for the clarification. I guess that means since the Naze32 uses an STM32_F1_03CB it requires an external inverter. Thanks!

Sleor, 2016-11-12:
Sorry to bump up this old topic.

I have flashed cleanflight 1.10.0 to my cc3d ( dont have a ftdi cable yet and the 1.10 is flashed with the .bin tru openpilot). But i have the exact same issue with Futaba sbus. I cannot get the cc3d connected with sbus enabled and battery powered. From usb it connects ok. So was this really fixed from cleanflight1.7 versions and up?

commit 3272b444 (2015-02-11): Update inverter.c

Fixes #494