cleanflight/cleanflight #498: Make MOTOR_STOP the default

danielgpalmer, 2015-02-12:
For people that have never used cleanflight(or multiwii) etc before having the motors spin up on arm can be surprising and potentially dangerous. Could the default out of the box be for MOTOR_STOP to be enabled so people that know what they are doing can turn it off and people that don't know what they're doing don't get a fright/cut their fingers off?

tracernz, 2015-02-12:
I would personally consider an armed craft which could potentially be bumped to full throttle much safer if the the props are spinning at extremely low power (which will stop if your fingers get in the way). At least you (and others who are potentially less knowledgeable about your craft) have some warning this way.

danielgpalmer, 2015-02-12:
I think it would go hand in hand with having the beeper beep when armed at zero throttle.

Making it default wouldn't stop you from disabling MOTOR_STOP to have the motors spin up when armed and would make it safer for people that don't realise that the MOTOR_STOP thing exists i.e. people coming from KK2. I suspect that a lot of people only realise that they probably want MOTOR_STOP after managing to get their FC to arm and have it unexpectedly spin up the motors.

If someone wants to show that it's not that dangerous with some sausages or something I would like to see that. :)

tracernz, 2015-02-12:
I am also not a fan of doing something because another project does it but each to their own. :stuck_out_tongue:

hydra, 2015-02-12:
I find myself in agreement with tracernz here, I feel its safer to spin slowly when armed.  Let's see who else casts their vote.

tricopterY, 2015-02-12:
I come from the MultiWii World and never appreciated the spinning props, when the craft is Armed. I am all for the MOTOR_STOP to be enabled by default, or alternatively reduce the MIN_THROTTLE default value from 1150 to 1050 (not sure if that would cause some ESC/Motor combo to hunt/stutter). 
The Armed LED status is often hidden from the view and is no help, so the spinning props win in that respect. I guess the documentation should clearly warn the user about the Arming Sequence and the configurator should also have a warning and disclaimer message on the Welcome Page.

ledvinap, 2015-02-12:
I have MOTOR_STOP disabled. IMO it is much safer.

fiveangle, 2015-02-12:
+1 on leaving current behavior. 

-=dave

RustyPaint, 2015-02-13:
+1 for spinning when armed by default

tricopterY, 2015-02-13:
Can someone please reduce min_throttle to say 1100 or 1050 for test, and see the spinning effect.
What if a user has a large rig and spinning 15" carbon props !!

ledvinap, 2015-02-13:
My ESC needs ~1080 to start spinning, so 1050 is too low ... Current default seems reasonable

tricopterY, 2015-02-13:
Thanks for confirming - the vote is to maintain the status-que.

nepalien, 2015-02-17:
+1 for keeping motor spin as default. Low speed spin coupled with the 5 sec Auto Disarm as default i feel is the safest option. Every cleanflight setup tutorial talks about motor_stop so any newbie should be aware of it and know how to disable it if they choose.

tracernz, 2015-02-17:
And once a setup wizard is implemented in the configurator this is a non issue.

https://github.com/cleanflight/cleanflight/issues/498#issuecomment-74608235

tricopterY, 2015-02-17:
If the aim is to curb accidental spin up once Armed, any Auto Disarm will not be a solution to an unsuspecting user - the damage would have been done. I personally enable Motor_STOP on the bench, before even installing the FC on a craft.

tracernz, 2015-02-17:
"Accidental spin up once Armed" doesn't make much sense. The whole idea of arming is that the motors are enabled. Maybe you mean accidental arming which is a different problem?

varesa, 2015-02-17:
I'm gonna leave another vote for the current behaviour. 

You shouldn't be handling an armed copter. The spinning props are a pretty good indication that the aircraft is "armed&dangerous".

Another way would be to have a continuous warning beep/tone when armed&still, but that would require additional hardware. The beep also might not communicate the danger level to some new users who decide to tweak an armed copter somehow. Props on the other hand...

tricopterY, 2015-02-17:
Based on personal experience, using MultiWii and having no knowledge of MOTOR_STOP
parameter in config.h file. I armed my first built multirotor (next to me on the bench with 11x4.7 props and 4S setup). You can guess the rest - curiosity killed the cat. Intentional arming, but to the observer Accidental Spin Up.
I never knew the motors would spin up. We live to learn. I guess this is why this thread was initiated. There is a lot of opposition to, having MOTOR_STOP enabled by default, citing a safer bet, if throttle is accidentally bumped to full. Looking at this objectively, there is a clear prerequisite -  Arm with the intention of Next Action i.e apply throttle. 
I am also against the option of arming, via Tx switch - that's madness.

varesa, 2015-02-17:
Maybe Ite would be good to have props off on a first time setup/arm :)

But this is kind of a two-edged sword: arm-time safety vs after-arm safety. Either way accidents will always happen either because of misknowledge, or purely by accident.

fiveangle, 2015-02-17:
"You can't make anything idiot proof because idiots are so ingenious." —Ron Burns

tracernz, 2015-02-17:
I propose to add a safety section to the manual immediately after the introduction. Something like this: https://github.com/tracernz/cleanflight/blob/safety-docs/docs/Safety.md . Thoughts?

tricopterY, 2015-02-17:
The proposed Safety and Warning Section is excellent and should have been there from day one.  There would be no problem, if everyone consulted the manuals and understood the terminologies etc. (think as an average user and not a developer). I wonder if motors could be Pulsed at MIN_THROTTLE, say at 2 second intervals, until the throttle input is greater than (MIN_THROTTLE + 50), then exit Pulse_MOTORS and enter normal operation. That should spin them up at reduced duty cycle, providing.visual cue, yet relatively safe. Scenario - Tx end-points/trims not set correctly, you can arm but cannot disarm. It happened to me, am I an idiot - most probably.

tricopterY, 2015-02-18:
Ok - I think we have been barking up the wrong tree. Already, there is logic to disarm the motors after a delay. My vote is to have the MOTOR_STOP enabled as default (auto_disarm_delay = 5 by default). This would be the safest for all users, and the argument of bumping the throttle by accident does not hold.
// When armed and motors aren't spinning, disarm board after delay so users without buzzer won't lose fingers.
    // mixTable constrains motor commands, so checking  throttleStatus is enough
    if (
        ARMING_FLAG(ARMED)
        && feature(FEATURE_MOTOR_STOP) && !STATE(FIXED_WING)
        && masterConfig.auto_disarm_delay != 0
        && isUsingSticksForArming()
    ) {
        if (throttleStatus == THROTTLE_LOW) {
            if ((int32_t)(disarmAt - millis()) < 0)  // delay is over
                mwDisarm();
        } else {
            disarmAt = millis() + masterConfig.auto_disarm_delay * 1000;   // extend delay
        }
    }

hydra, 2015-02-18:
I think @tricoptery has a good point there. You would have to bump the throttle within the armed period and its likely you're already holding the tx after arming so you already know you just armed it.

With that in mind I am now in favour of changing the code so motor stop is enabled by default.

What do you all think?

ledvinap, 2015-02-18:
Personally I prefer spinning motors when armed. It forces you to respect armed state, no exceptions. Auto disarm seems annoying, sometimes it is reasonable to wait on ground when armed.

ienders13, 2015-02-18:
I thought the same but now have motor stop on so that I can kill the motors on landing just by throttle off which will be the best scenario when I dump my 250 racer in long grass or a bush or what ever I know I can kill the motors quick and not risk overloading anything. Auto disarm makes sure you don't forget you've armed it and spin up by accident as 5 seconds goes quick. I think it works better as arming and flying is a 2 stage process and less prone to accidental spinning up the props.

fiveangle, 2015-02-19:
The truth remains that unless a buzzer exists, or the flight control board has an LED to show armed state (and the flight controller is not buried in the chassis), the user has telemetry, OSD, or some other indicator exists, there is no indicator that the craft is armed. The new "auto disarm" feature only re-enforces the need for positive indicator of armed state for otherwise new users who might otherwise be extremely confused at the inability to arm the craft… until they are successful after building up a false sense of security that the blades will never spin up.

Strip away all the custom modes, led strips, OLED displays, telemetry, OSD, and other stuff - MOTOR_STOP makes sense and should be held up as a first rite of passage for newbies to deactivate once they understand and respect the "armed" state for what it really is: dangerous

-=dave

matvey83, 2015-02-26:
I initially thought I wouldn't like it, but after a few flights I actually like the current behavior of having the motors spin slowly until I disarm.

(One thing that is confusing, IMO is the name of this feature: MOTOR_STOP)

2 cents

lamnesique, 2015-02-27:
hi, 
personnaly i set all my quads with the feature that the motors don't spin when armed for 3 reasons : -when i armed i know what i want and what i do so i don't need to have spin
-when i do some setup with just armed/disarmed on desktop it's just idiot and dangerous that the motor have to spin..
-when i crash (yes i'm not perfect like many pilots) in long grass or others it's stupid to burn anything or to brake props because the motor spin if you forget for sometime to disarmed

but i'd like the idea to add a feature that when we have buzzer, when we armed it "long bip" one time to advise people around that don't see the danger that a quad is preparing for take off

all is just my opinion and a little idea

mosh1, 2015-02-28:
Having motors spin up when armed was one of the things that surprised me the most with Baseflight and looking to possibly switch to Cleanflight today, this is one of the first things I checked to see if it was fixed. Through years of R/C and dozens of models and electronics of all sorts this was the only time I've experienced motor throttling up on arm and where the throttle signal on control is zero. So this issue is dangerous for both newbies and experienced R/Cers. Reasons for MOTOR_STOP to be enabled:

1). Craft being armed is an explicit action. You're already at the controls and on alert. Unlikely to bump throttle.

2). Most R/Cers have a throttle hold switch set up and learn to trust it. This is an additional safety layer over flight control arm/disarm. This switch keeps throttle to zero no matter what actual stick position is. This prevents accidental bump throttling, and also gives a quick emergency off that's learned by muscle reflex if aircraft is crashed, or you need to quickly dump the aircraft doing to some danger (human or dog in way).

3). Motor spinning on arm removes the emergency safety feature of #2. A few mentioned an emergency off when crashing already, but more important is when safety of other objects of concern. Going through a slow disarm sequence isn't an option. You need to be able to jam throttle stick down or flick the throttle hold switch. 

4). There is no such thing as a 'safe slow speed' as others mentioned. This may be true for certain size props and certain motor kv / voltage combinations. But not true when you're talking large carbon props and high kv/voltage setups. 

It was very surprising to have my motors spin up (props off) when my throttle hold switch was enabled and I armed the flight controller. Users who would rather have that behavior can still set it manually (although for safety reasons I listed above it seems silly) - but I think default of MOTOR_STOP is much safer.

tricopterY, 2015-02-28:
+1 all the way. It's crazy to have motors spin when armed. @mosh1 explained it really well for most scenarios/users. You smart people, who prefer a spinning prop - think about dumb guys like me too. You can disable MOTOR_STOP to suit yourself, at will.

mosh1, 2015-02-28:
IMO the "smart people" should still have MOTOR_STOP enabled, and instead if they want spinning props at zero throttle when armed to set up a separate flight mode on the radio for this that adjusts the actual throttle curve on the radio for that flight mode to suit. In this case that flight mode can also be an 'aerobatic mode' where they won't have the motor completely stop at lowest throttle stick if they are doing flips and rolls.

In an emergency, this will still let the throttle hold switch to work. And if you're working on the aircraft you can be sure that with throttle hold enabled on your radio even an accidental arm or bump won't spin the props. Now you get the best of both worlds and still retain the safety.

fiveangle, 2015-02-28:
#1 I think the problem here is that it's *not* explicit, because it's performed by relatively "loose" stick movements and is prone to not engaging under many circumstances when setting up a new craft, so if someone performs the arm maneuver and the craft doesn't immediately respond to throttle up afterward, then it is in an unknown state (TX channels not set up correctly, THR not working at all, accidentally armed but not realizing it, etc). This is the condition that I think is feared. If everyone had multi channel transmitters, we probably wouldn't have arming by sticks at all. 

#2 not all radios have this capability. If they did, this feature would have likely never been implemented. Burning an aux Channel for arm (which can be added as a mix in the tx) exists today for those that wish. Not all have that capability, which I think is the real issue here. If everyone could look down at their transmitter and see a stick in the THold position, then I don't think we would have a problem. 

#3 not if its setup correctly. I can show you on my setup next time we're out. If your argument is "most RC flyers are used to THold for safety so we should use that as a baseline default" that merits discussion, but so far it doesn't seem like a strong relationship.  Again, if a single RC channel directly controlled the esc/motors, I don't think this feature would exist, but that the flight controller truly has final say, what's coming in on the throttle line really is somewhat arbitrary compared to the ARM function of the controller. 

Anecdotally, the biggest problem I've encountered was accidentally arming the craft when I meant to enter the MWOSD gui but had THold enabled, which seems worse imho, again, because the craft was in an unknown state to me, the user. Because I knowingly enabled the MOTOR_STOP feature, it required investigation on why it didn't behave as expected (which is when I would find TX THold was on, so the stick movement to enter the OSD translated into the stick movement to arm the craft because throttle was at zero the whole time). Luckily, my example was the safe outcome with THold complementing the accidental arm. But the confusion was real and imho, should be avoided if at all possible, which this feature does.

#4 has real merit but I think the expectation is that the likelihood of someone building a huge carbon prop bohemouth as their first foray into this is near zero chance. Nonetheless you have a very valid point.I've not heard of such horror stories but I'm sure they must exist :)

Adding to the confusion, with the auto disarm feature, there is no positive method to determine if armed without the addition of a beeper or visual LED or some other indicator. There have been reports of users not figuring out how to arm when in fact they were armed but then not using throttle before auto expiration (obviously with MOTOR_STOP enabled). I could just imagine someone confused spiraling into a flail on the sticks until suddenly it spins up while they're holding it in their hands shaking it. Btw- I personally love the auto-disarm feature, but I do admit that it adds a layer of protection that makes it easy to rely on, which isn't necessarily a good thing. The new user confusion example is also worrisome. 

There are good arguments on both sides, with either option being best for certain contexts, which usually means it's best to have less debate and more autocratic decision-making by those in power. We'll live with the outcome either way :)

-=dave

mosh1, 2015-02-28:
#1 For the specific situation of those setting up a new craft, there can be much better ways of letting the user know it's armed than spinning the motors. Flashing lights, audio and an alert when connected to the configurator, etc. And if those won't work then they can explicitly disable MOTOR_STOP.

#2 I don't think it's common at all today to have a radio that isn't capable of using an aux for a throttle hold. Haven't checked. But if so, then it's even more important they can actually bring props to a full stop with a throttle stick down on an emergency without having a switch or throttle hold programming capability.

#3. If you mean, you set up a switch to quickly arm/disarm instead of slower stick movements - this isn't ideal because either you lose the second safety of a throttle hold or you have to switch two switches (one for arm/disarm another for throttle hold) in an emergency. A one switch arm or throttle hold was fine for gas r/c ages ago but not IMO not safe enough for electrics today. This is why stick commands are actually safer for arming - less likely to be bumped, and a second throttle hold switch even safer on top. 

#4. Someone building a huge prop quad will likely have a lot of experience in r/c and perhaps even other boards with different flight control software. They may be experienced enough that they come with an even larger expectation that when they arm their quad for the first time the props should not spin. Cleanflight/Baseflight is the exception - not just with open source quad specific flight controllers but of all flight controller types i've ever personally seen and used.

Since this issue is just for what should be the better default - it should be focused on safety and expectations for new and experienced users coming into Cleanflight for the first time. If someone still thinks having the motors spin is the best way for them to know their aircraft is armed, they do it explicitly and will have no surprises.

tricopterY, 2015-02-28:
@hydra - please make an "autocratic decision" and make CF a much safer bet for all. People who have MOTOR_STOP disabled, never Arm their FC when on the bench or close to them - they know the motors will spin. So, what's the concern? Do they really need to see spinning props, to give them a visual indication - that's clumsy?

makattack, 2015-03-27:
I've always viewed the motor_stop default for multiwii and the forks as almost an homage to internal combustion powered models. Still, even with gas helis, there is a clutch for throttle hold/auto rotate. It seems to me unnecessary for electric powered aircraft. I've always disabled it since multiwii 2.x.

I also use an aux  channel to arm my tricopter top avoid vibrations traveling down the booms, but also want the servo active while disarmed so I can function test it before arming.

hydra, 2015-05-15:
A new problem surfaces, if i enable MOTOR_STOP by default along with the new pid_at_minimum_throttle (which is current enabled by default) then if you arm the motors will spin if the craft isn't quite level.

Based on this, i am inclined to leave MOTOR_STOP disabled until I have more feedback.

Futher comments, feedback and suggestions welcomed.

mudcamper, 2015-07-17:
Let me quote mosh 1, because he said it best:

2). Most R/Cers have a throttle hold switch set up and learn to trust it. This is an additional safety layer over flight control arm/disarm. This switch keeps throttle to zero no matter what actual stick position is. This prevents accidental bump throttling, and also gives a quick emergency off that's learned by muscle reflex if aircraft is crashed, or you need to quickly dump the aircraft doing to some danger (human or dog in way).

3). Motor spinning on arm removes the emergency safety feature of #2. A few mentioned an emergency off when crashing already, but more important is when safety of other objects of concern. Going through a slow disarm sequence isn't an option. You need to be able to jam throttle stick down or flick the throttle hold switch. 

This is the truth. For any experienced RCer, particularly if we predate the entire multi-rotor scene, having a throttle kill is the first and arguably most important thing to set up on any new model. This poor default behavior on CleanFlight breaks this. On every model I own, I can hit the same switch on my transmitter, and make it safe. Until now. Now CleanFlight has broken this. Yes, I can go change the default behavior. And I will. But let me describe my introduction to it for you, which will be similar for many other experience RC pilots:

I power up my new multi-rotor for the first time. I arm the flight controller (left stick down-right). To my surprise, the props spin up. I hit my throttle kill. Nothing. I tell the flight controller to disarm (left stick down-right). NOTHING. Nothing I do disarms the flight controller. I reach my hands in between spinning blades to disconnect the battery. 

Why did this happen? First, the aforementioned stupid default to power up the motors with the throttle off. Second, I hit my throttle kill, which set the throttle position LOWER than the lowest stick position. This in turn confused the left-stick-lower-left-to-disarm code. Once I throw my throttle kill switch, it crashes the disarm ability in the flight controller, even if I turn the switch off. Yes, I can change the relative off position in my switch, and change the default motor_stop, but this is all poorly designed.

The default should be to STOP THE PROPS from spinning when the throttle goes to the off position.Anything else is utterly nuts.

Trussroads, 2015-07-17:
This is also related to and an example of the fact that cleanflight is becoming unnecessarily complex to setup and to achieve the best from. Also highlighted by the fact that so many of the code writers themselves have their own personal version that they run and patch update as features are realised in the main stream.

I do personally feel there is such a focus on speed of development that some fundamental "cleanness" of use and function such as this seems to be missing? 

Dominic has done such a great thing here and we have some of the very best coders/pilots on the project that it remains one of the most promising projects underway.. I really hope to see it settle into a solid and very robust platform and am making these comments in the hope of constructive encouragement only.

matvey83, 2015-07-17:
For those whose Throttle Hold isn't turning off the props:

The proper way to set up Throttle Hold with clean flight is to hook it up to the ARM channel.  When TH is activated, the model should become disarmed, stopping all motors.

ProDrone, 2015-07-19:
Experienced RCer __never arm__ a new multi-rotor __for the first time with props on__, but double checks everything before even thinking about mounting the props.

@hydra: i think it is simple, for inexperienced users we should put safety first. The safest you can get is change defaults for both: MOTOR_STOP default enabled (props not spinning on arm) __and__ `pid_at_minimum_throttle` default disabled. Users need to read the manual to configure it (deliberately) to a less safer preference...

mosh1, 2015-07-20:
@matvey83 I disagree that's the proper way. Electrics ideally should have a non toggle switch arm. I've seen these happen frequently over many years in R/C: you can accidentally leave the arm switch on, you accidentally bump it arm (sometimes when the throttle is high), your radio getting moved around arms it. Ideally arming isn't a toggle but an explicit action and disarm can be toggled. This is why I use both a toggle throttle hold that keeps throttle below 0% and arm via explicit stick actions.

In any case this whole issue is about what default should ship with cleanflight and I think most would agree MOTOR_STOP enabled is safer.

joshuabardwell, 2015-07-20:
>In any case this whole issue is about what default should ship with cleanflight and I think most would agree MOTOR_STOP enabled is safer.

Are you sure? Maybe we should set up a poll. I firmly believe that MOTOR_STOP is less safe, because it can leave the quad armed without an obvious sign that it is armed. MOTOR_STOP just encourages unsafe behavior by making it easier to handle the copter when it is armed.

JohnOCFII, 2015-07-20:
I agree with Joshua. I like to see the props spinning on arm, as I know we are in a "live" situation.

danielgpalmer, 2015-07-20:
@joshuabardwell

The default settings seem to disarm after 5s with no throttle. So in most cases if you are handling the the model when it's armed it's going to be disarmed pretty quickly.

joshuabardwell, 2015-07-20:
>The default settings seem to disarm after 5s with no throttle. So in most cases if you are handling the the model when it's armed it's going to be disarmed pretty quickly.

This is true whether motor_stop is enabled or not, right? So disabling motor_stop still enhances safety during the window between arm and automatic disarm. The exact same arguments apply, as long as the automatic disarm applies to both settings.

mosh1, 2015-07-20:
>  I firmly believe that MOTOR_STOP is less safe, because it can leave the quad armed without an obvious sign that it is armed.

I think for those that feel that way they set it explicitly.

By default, cleanflight is set to disarm by stick positions and also set to default to spin the motors when armed. By default there is no way to quickly kill the throttle when either the user does not expect the motors to spin up when armed, or in a danger situation (eg. hitting someone). This alone I feel merits default to MOTOR_STOP.

As mentioned, if you set disarm to a toggle switch instead of stick positions, then you can kill motors quickly. But this isn't the default with cleanflight. Also many experienced RCers want an explicit throttle hold (that holds throttle under 0%) while maintaining the stick arming since as I mentioned previously - toggle switch arm/disarm is unsafe for electrics because a toggle is too easily left enabled or bumped.

Also for someone mentioning it will disarm after 5s. 5s is an eternity when you're being cut by blades. For small models this isn't an issue but many fly high torque powerful setups. I base my experience from dealing with very large electric helicopters.

ProDrone, 2015-07-21:
@mosh1 Can you please review #1149 and give your input on this? I personally have motors spinning when armed because of the visual indication of danger, i wish there was another method that worked on all copters to achieve the same...

ProDrone, 2015-07-21:
> This is true whether motor_stop is enabled or not, right? So disabling motor_stop still enhances safety during the window between arm and automatic disarm. The exact same arguments apply, as long as the automatic disarm applies to both settings.

Automatic disarm only works when feature `MOTOR_STOP` is enabled AND throttle is low.

fakatwork, 2015-07-21:
For what it's worth I vote to keep the current behavior with motor_stop = off for the same reasons(s) the other people evoked.

I have arming on a AUX switch that I also use as a kill switch when crashing, and it's not risky as some implied because if I bump the switch by accident while the throttle is not zero it won't arm anyways.

bdoiron74, 2015-07-22:
@fakatwork Ditto.

mosh1, 2015-07-22:
Why have the motors spin on arm by default though? For those that want that as an indicator of arm, why not set it explicitly so you know to expect that behavior? Or some might rather use lights or a beeper to indicate arm, but for the person new and installing cleanflight for the first time spinning up on arm is a pretty big surprise and why I vote it's dangerous.

ProDrone, 2015-07-22:
_It is only dangerous with props mounted_.

But these should __not be mounted__ when setting up and testing a new multi-rotor, but only _after configuration  & verification_ of the transmitter, receiver and Flight Controller.

This is a safety measure and something that beginners should be aware of. Maybe we can extend this document https://github.com/cleanflight/cleanflight/blob/master/docs/Safety.md and put a direct link (near the top) in the readme.md?

Spinning motors to indicate successful arming is helpful in the setup phase. If one prefers to have `MOTOR_STOP` enabled, it is best to change that as a last configuration change.

When having a model with known behavior on the field things become different. I think it is a good thing to make arming a procedure (flipping a switch followed by stick arming) so that it does not happen by accident and make disarming a direct action (kill switch) to prevent damage in case of emergency. This is what i want to achieve with PR #1149

If you rely on settings in your transmitter, also consider what happens when it is (accidentally) OFF or what happens under fail-safe conditions.

closed 2017-02-25 by hydra