cleanflight/cleanflight #503: CF config, needs a check box (for each FC board) to pick your board

Ozzy33, 2015-02-12:
In the firmware flasher, please add a check box (for each FC board) to pick your board, then it only shows FW for your board, then in the drop down, it will look cleaner. TIA

thenickdude, 2015-02-12:
That would trade off extra dropdown entries for a big honking array of
tickboxes, this doesn't sound like it would be cleaner overall.

tracernz, 2015-02-12:
The best I could see from this is a separate dropdowns to select target and version.

tricopterY, 2015-02-12:
Currently the code simply populates the dropdown by getting all it sees in github cleanflight/releases url. It looks messy with terse parsing and formatting. My vote would be for separate Target Dropdown -options hard coded with supported targets, and on target selection populate a Listbox with related versions, fetched/filtered from releases url, with latest version selected.

hydra, 2015-02-12:
It's possible now for the configurator to ask the board what it is if it already has cleanflight on it, so this becomes less of an issue when re-flashing.  For the first-time flash it's much more important the user chooses the right thing.

the drop-down box WILL go, it's a terrible choice since it does not work when there is more than a few items to choose from.  It is a legacy from the fork of baseflight-configurator.

I was thinking of a paginated table all releases, filterable by board and release status (e.g. protentially hide-release candidates by default) with release notes in a column.

I was thinking of displaying pictures of the boards in the UI too.

AlwaysTinkering, 2015-02-12:
Slightly related, is that the page height for the flash tool is getting long enough that for most people, if the flash button at the bottom is visible, the progress bar for the flash at the top is scrolled out of view.  This is somewhat dependent on the length of the detail section for the selected firmware that appears once a dropdown item is selected.

This leads many people to not realize the flash was successful.

tricopterY, 2015-02-13:
How about a two level Menu:

[image]

tricopterY, 2015-02-13:
How about a two level Menu eg.

[image]

hydra, 2015-02-14:
Eww no. Just gimme a little time guys and I'll show you the solution.

hydra, 2015-02-14:
@AlwaysTinkering yes I will address that too.

tricopterY, 2015-02-14:
@hydra : IMHO - please aim for a less ambitious solution. A paginated table would add to confusion, with older releases cluttering the page. Most users want the latest release and the option of seeing a candidate.
A link to the release repository would assist those curious people, looking for candidate release, as well as older versions, that can be downloaded as hex/bin. eg.
[image]
A picture of the FC, is unnecessary effort for bells and whistles. There are too many Naze clones, as it is. My two cents worth.

heathicus, 2015-03-02:
If a change has to be made, why not just separate the Firmware and Board options into two drop-downs.  First drop down chooses a board.  Then the second drop down is a filtered list of the firmware applicable to that board with the most recent stable being the default.

hydra, 2015-03-02:
that is closer to the mark @heathicus yes.

Using drop-down menus and  sub menus is also not scalable and would make (IMHO) a horrid user experience.

providing the 'board' as a dropdown is fine, since there is a small list that changes infrequently and would lend itself to being automatically set based on an MSP inquiry to the board.

the main issue to fix is the dropdown, it's not a scalable ui element.  It's likely we could just show the last 10 releases (per target) in a table and provide a link to the github site of a user wants an older one.

using a table to show date, version, release status and a couple of lines from the changelog (followed by "...") which expands when selected is my preferred solution.

Agree that pictures are not needed for first cut of ui that doesn't use a select box for the firmware version, that is a 'would be nice' feature.

tricopterY, 2015-03-04:
I have done a rush hack on the firmware flasher tab (needs better css styling) - that can be downloaded from:
https://github.com/tricopterY/cleanflight-configurator

[image]
The AlienWiiF1 does not work, as it does not find a match, got do all lowercase compare etc.

tricopterY, 2015-03-04:
My rushed hack of Firmware Flasher - https://github.com/tricopterY/cleanflight-configurator
[image]

tricopterY, 2015-03-08:
I would appreciate a feedback for the firmware flasher hack. Just download it and test it in the interim, until @hydra does his proposed paginated table.
The only side effect is for CJMCU, only gets the last three, since one in-between release is missing from the traversed indexes.
I have also updated the change log, with my fixes.

Ozzy33, 2015-03-10:
Another possible solution would be each FC board has it's own text color in the drop down box.

tricopterY, 2015-03-10:
This issue has been exhausted and the tacit vote is for a paginated table UI, that would handle all releases for each supported FC board. I personally prefer only the latest three or four releases shown, the rest are historical and serve no real value to a user.

Ozzy33, 2015-03-10:
I agree to a point, what If I have a stellar PID setup for an older FW, I should be able to select it and dump my old settings in and go, not re-tune. it's all good, thanks CF for all the fixes and improvements!!!

tricopterY, 2015-03-10:
You could simply get that Older Release from 'CleanFlight Release Repository' and download it to your PC. In Firmware Flasher you would use Load Firmware [Local] button to select the hex file from directory you downloaded it to, and then flash the FC. I am not in favour of cluttered single screen UIs - most options/views don't even looked at, yet alone used effectively. Happy flying.

Ozzy33, 2015-03-10:
LOL, like I am? I started this request,,,,

heathicus, 2015-03-10:
I agree with tricopterY that the list of firmware releases should be limited.  No need to list all of them, or even 10.  The current stable release (the default option), the current release candidate or experimental or beta or whatever you want to call it, then 1 or *maybe* 2 older stable releases.  Then a link somewhere on the screen to a repository of older releases.  Simple and easy.

Roeland54, 2015-03-10:
Maybe the solution is simple. If the list would only show firmware versions that are supported by the installed configurator version it would be a lot shorter. 
If you want a older firmware then install the older configurator and you can flash the older firmware...

What is the point of flashing version 1.2.0 if the configurator doesn't support it?

tricopterY, 2015-03-10:
+1 to @heathicus and @Roeland54  - scenarios very well articulated.

tricopterY, 2015-03-11:
Inviting discussion for a simple UI - and not concerned about scaling, as it would only fetch the latest four releases from CF Rel Repo. Already working great for me - the get mechanism remains as is, and rel notes can be read when the hex file is fetched. If I need an older version I would click on Release Repository menu item and it would take me there.
[image]

tricopterY, 2015-03-11:
Inviting discussion for a simple UI - and not concerned about scaling, as it would only fetch the latest four releases from CF Rel Repo. Already working great for me - the get mechanism remains as is, and rel notes can be read when the hex file is fetched. If I ever needed an older version I would click on Release Repository menu item and it would take me there. I am unable to get a closer match to the colour of CF logo etc.
[image]

hydra, 2015-05-15:
Closing, this is a configurator issue.  If an issue doesn't exist in the configurator issue tracker please create one and reference this.

closed 2015-05-15 by hydra

AlwaysTinkering, 2015-11-01:
Hey!

 

New message, please read <http://americantrailermart.com/felt.php?xm2a>

 

bert@drakedev.com

AlwaysTinkering, 2015-12-17:
Hey!

 

Open message <http://belladonna-pa.com/happy.php?rq9>

 

bert@drakedev.com