cleanflight/cleanflight #505: Missing documentation for cli variables/commands (i.e. motor_pwm_rate)

skyflyer, 2015-02-12:
I looked through the documentation and could not find documentation for CLI variables. I found the documentation on differences between Baseflight and Cleanflight, but I guess a complete command reference would be welcome.

For instance, does motor_pwm_rate higher than 500 (i.e. 501 or 600) really mean that it will drive the ESC in brushed mode? What would that mean in practice? Related to this is issue #58.

For startes, I guess one could copy/paste the baseflight cli variable documentation and then improve on it.

tracernz, 2015-02-17:
I had a quick look and came up with this https://github.com/tracernz/cleanflight/blob/cli-docs/docs/Cli.md, I'll work on populating the descriptions.A Python script which parses serial_cli.h and another file containing descriptions, then puts it all together and auto generates the table would be nice for ease of updating in the future..

skyflyer, 2015-02-17:
It is a good start. With regards to python script: is there something similar in place? If it is, let me know, if it isn't, I can take a shot at it and write a simple parser for documentation. `serial_cli.h` is empty, but `serial_cli.c` has some stuff in it, although not easily parsable. Do you think it would make sense to add structured comments for those commands? Or use something like sphinx & doxygen to generate documentation from comments. This sounds overly complicated for starters, so we could start with a simple script, but still annotate comments in the code with directives for parsing.
Let me know if I can help.

tracernz, 2015-02-17:
I'll work on it in the next few days and pull some descriptions together. Getting a manually generated table together will be a good start. :smile:

waltr1, 2015-02-17:
Documenting all the CLI variables would be great. Currently one needs to read the BaseFlight CLI document for the one that are now in the CleanFlight CLI document.

hydra, 2015-02-17:
Please submit a pull request to start this process.

tracernz, 2015-02-17:
I'll get some descriptions from baseflight docs, code etc. and then submit a PR very soon. :smile:

tracernz, 2015-02-20:
@hydra, #520

waltr1, 2015-02-21:
The CLI descriptions in the new doc look good.
I would like to suggest some notes added that refers to other docs for details.
Fail Safe, OneShot, Serial, Mixer, etc.

tracernz, 2015-02-22:
@waltr1, most definitely. The initial PR was just to get the ball rolling with something rather than nothing. Feel free to come up with some ammended and/or additional descriptions and we can get them included.

hydra, 2015-03-03:
Ok, closing this now since the PR was merged, please submit new PR to update the documentation as required.

closed 2015-03-03 by hydra