cleanflight/cleanflight #506: Drift in PID Controller 5 (harakiri)

nebbian, 2015-02-13:
Several people (myself included) have noticed that when you switch to PID Controller 5, there is a noticeable drift when in acro mode.  It's as if the midpoints of the RC channels, or the gyro calibration is not being read correctly.

All other PID Controllers are fine, it's only PID Controller 5 that shows the issue.

MJ666, 2015-02-13:
There must be some additional condition in combination with PID 5 which causes the drift. Actually I fly 5 different Nanocopters without having drift problems. Details can be found herer: http://www.rcgroups.com/forums/showpost.php?p=30765614&postcount=2894
There also some others who fly PID 5 quite succesful: http://www.rcgroups.com/forums/showpost.php?p=30763112&postcount=2871
Flying like this will not work with a drift. We need to find the condition why this is happening.

borisbstyle, 2015-02-13:
@MJ666 
Yes I saw that video of bird of prey and I am really surprised it worked out for him, but it didn't for me. 
It would be nice to see his pids and settings he used and compare with mine.

My integral was raised to around 0,100 and still no noticaeble difference or perhaps need to use even higher numbers?

HFMan, 2015-02-14:
I also found that on my Nemesis, in Acro mode the quad still seems to want to level out if you let off the sticks.  It's a slow level, but it does it in any event.  I can't tell if it's due to a complete lack of 'I', or it actually is somehow using the accelerometers.  This controller to me is a bit scary- I still haven't gotten comfortable with it.

ledvinap, 2015-02-14:
@HFMan Just guessing, but if COG is below propellers, gravity will try to level the quad.

Crashpilot1000, 2015-02-18:
Had a quick glance at the port here in Cleanflight, as far as I can see the "I" term calculation is mostly wrecked during porting because int32_t variables are used and can not contain fractions - that's why I use floatpointvariables for that in Harakiri...

hydra, 2015-02-18:
Any chance of a pull request to fix the drift then @crashpilot1000 ?  We can address the use of unprocessed gyro data and mixing as separate tasks.

nebbian, 2015-02-18:
Thanks for the heads up @Crashpilot1000.  I've made the changes that I think you're talking about, and flown them.  I can verify that I couldn't notice any drift in Harakiri (compared to the other controllers) with the new code in place, whereas it was quite noticeable previously.

hydra, 2015-02-18:
Can you make a PR for this asap so I can include it in 1.7.2

nebbian, 2015-02-18:
@hydra the pull request is this one: https://github.com/cleanflight/cleanflight/pull/522 (also shown three comments up)

MJ666, 2015-02-18:
@Crashpilot1000 good catch looks to be I missed this. @nebbian thanks for fixing this.
I could even not see the drift on my Nano's and was flying PID 5 most of the time.

Crashpilot1000, 2015-02-18:
@hydra: I will try to look into that but like you stated in the video I am not a programmer.
@nebbian: Thanks for the quick fix! I scooped through your fix and there still seems to be something "odd" since the "errorGyroI_YW" is int32 and the angle mode missing float in I-term accumulation is not addressed. So ratemode fixed, angle mode untouched and yaw probably injected with an error on the I-term. I know my code is a pain but here is the link for reference: https://github.com/Crashpilot1000/TestCode3/blob/master/src/mw.c .

MJ666, 2015-02-18:
I understand the problem now and made some comments in the PR.

nebbian, 2015-02-19:
Thanks @Crashpilot1000, I've had a look at your original code and see what's going on.  I'll address this in about 12 hours.

Crashpilot1000, 2015-02-19:
Yep, nebbian you probably wouldn't notice it in flight but Alex put a lot of work into that yaw controller for multiwii "yaw ITerm windup is very high, allowing an 'elastic' direction return after a manual perturbation" so...

MJ666, 2015-02-19:
@all finalize the work on PR #526 may be some of the pilots who had the drift problems could verify if this is resolved. I flight tested the new firmware on two Nanocopters and could not see problems. But remember I cold not se the drift problem here.

hydra, 2015-02-19:
I will merge this now.

@Crashpilot1000 Thanks for helping out on this - *really* appreciated!  Please let us know if you find other issues and/or have other suggestions.
@MJ666  @nebbian thanks also for your great contributions.

hydra, 2015-02-19:
#526 merged.

closed 2015-02-19 by hydra