cleanflight/cleanflight #512: Support for the MS5607 barometer

jmorgan1, 2015-02-14:
Cleanflight already has support for the MS5611 barometer which comes populated on the full Naze32 board, but I did not see any support for the MS5607 which is exactly the same as the MS5611 but with half the resolution. Creating drivers for it would be very simple and just require changing a few conversion factors when calculating the temperature compensated pressure from the device.
I ask for this enhancement because I recently purchased an Acro Naze32 board and I sourced the MS5607 barometer myself and soldered it onto the board after I received it. It works, but the calculated values are wrong since Cleanflight is using the driver for the MS5611 sensor.
I am unsure yet if it would be possible to automatically detect which type of barometer is installed on the board in order to use the correct driver, but if not then the user might just have to manually select it.

jmorgan1, 2015-02-14:
Relates to pull request #513

hydra, 2015-02-17:
Great, i will have a look at that,  there's quite a backlog at the moment.

ledvinap, 2015-02-17:
Is new driver really necessary? Only few pressure calculation needs to be changed, everything other is identical to MS5611. So IMO it would be better to add flag to MS5611 driver to use different math. And maybe rename it to ms56xx.

Autodetection is probably possible using PROM register 0 or 7, but it's content is not documented (or my google-fu is too low)

jackm, 2015-02-17:
The existing drivers having the name "ms5611" was one of the main reasons I chose to just make new but similar files. If the driver was just renamed to ms56xx then I suppose that works too.
I haven't looked into what the PROM values are in the ms5607 vs the ms5611 though. If needed I could probably gather what information is on the ms5607 but I don't have a board with the ms5611 to compare that to.

ledvinap, 2015-02-17:
I can check data from MS5611, but data may be dependent on revision. It may make sense to use your values now and add new signatures if some other user needs them. 

Duplicating code is IMO bad idea.

@hydra - what is your position on this? (CLI variable/autodetection?; ms5611 rename?)

hydra, 2015-02-17:
ms56xx I think is better in this case, especially when coupled with a unit test that checks the  calculation for both sensors.  2 birds one stone.

jackm, 2015-02-17:
Alright I can do that. Shall I make a new pull request after and throw out the existing request?

jackm, 2015-02-17:
I've made the changes in a new branch of mine: [jackm:ms5607_support](https://github.com/jackm/cleanflight/tree/ms5607_support)

This still does not have the auto detection in it to decide which sensor model is being used. I couldn't find much information about the PROM data content for these sensors, but looking at the first word in the PROM would be a good start since the datasheet says "Address 0 contains factory data and the setup".

ledvinap, 2015-02-17:
https://github.com/jackm/cleanflight/compare/ms5607_support#diff-70e415be8ca58e30823ca50d45c7d07fR181

Cast to int64_t is necessary here ..

jackm, 2015-02-17:
Like such?: `temp -= ((int64_t)dT * (int64_t)dT) >> 31;`

ledvinap, 2015-02-18:
Yes. Technically, it is enough to cast only one operand, but this will work.

closed 2017-02-25 by hydra