cleanflight/cleanflight #514: Allow the number PWM RC channels in use to be configured.

danielgpalmer, 2015-02-15:
While upgrading from baseflight I noticed that the variable power_adc_channel is not present. This is used in baseflight to move the current sensor input to one of the RC inputs that isn't in use.
At the moment it seems that cleanflight won't allow me to enable the current meter feature because I am using PWM input which would conflict if I was using 8 channels but I'm only using 6.

If cleanflight allowed the user to configure how many PWM inputs they are using then it would allow for pins that have other features to be enabled with PWM input.

hydra, 2015-02-17:
Yes, pin allocation is something we want to cleanup.

See #357

hydra, 2015-05-15:
See #884

closed 2017-02-25 by hydra