cleanflight/cleanflight #518: AUX Mode values now wrong in the mode.md document

waltr1, 2015-02-17:
I could not get the beeper working from a TX Aux switch. Posting on the RCG CleanFlight forum resulted in tracernz replying with a copy/paste from the source code.
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/rc_controls.h

[code] typedef enum {
    BOXARM = 0,
    BOXANGLE,
    BOXHORIZON,
    BOXBARO,
    // BOXVARIO,
    BOXMAG,
    BOXHEADFREE,
    BOXHEADADJ,
    BOXCAMSTAB,
    BOXCAMTRIG,
    BOXGPSHOME,
    BOXGPSHOLD,
    BOXPASSTHRU,
    BOXBEEPERON,
    BOXLEDMAX,
    BOXLEDLOW,
    BOXLLIGHTS,
    BOXCALIB,
    BOXGOV,
    BOXOSD,
    BOXTELEMETRY,
    BOXAUTOTUNE,
    BOXSONAR,
    CHECKBOX_ITEM_COUNT
} boxId_e;[/code]

By commenting out: " // BOXVARIO," this broke the document and now all the AUX Mode values below  BOXVARIO needs to be decreased by one.
I think this is very bad coding which can not only break other pieces of code but makes documenting the values impossible. This should have been filled with a 'dummy' name to maintain the structure.

hydra, 2015-02-17:
See #344

hydra, 2015-02-17:
What program are you using to configure the mode?  Cleanflight configurator or something else?  Which version of firmware and configurator?

hydra, 2015-02-17:
Also, the list of modes is decoupled from the msp commands, so your comment about bad coding is incorrect.

See:

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L304-L329

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L476-L532

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L936-L948

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L598-L656

Also, if it didn't work it would mean the configurator wouldn't be able to assign modes to anything following the disabled BOXVARIO value, like BOXCAMSTAB and I was using this exact feature myself a couple of days ago and it was fine.

waltr1, 2015-02-17:
I use the CLI and the Documents in github. So these are not matching.
Having the Docs not match the code and then kludging the configurator is very bad programming practice. This causes huge confusions which will bit uses and future programmers.

tracernz, 2015-02-17:
The MSP commands are correctly tied to the permanent box IDs but the setting the CLI variable alters the variable directly which relates to the enum in https://github.com/cleanflight/cleanflight/blob/master/src/main/io/rc_controls.h#L22-L47. I have fixed this enum to align with the MSP permanent IDs in #510.

hydra, 2015-02-20:
which cli commands are you referring to?

hydra, 2015-02-20:
@waltr1 with regards to your comments regarding bad programming and kludges.  Please be more aware of the history project when making such statements, there is much legacy code in the configuration and firmware due to the initial implementation of the MSP and MultiWii Firmware.

It would be also more constructive to point out exactly where the issues are in the code, links to specific line numbers go a long way here. Suggestions on how to fix and statements of fact are most welcome if you want to get things fixed.

waltr1, 2015-02-21:
Ok Hydra but I did explain above.
I posted the boxId_e enum that show the one mode commented out in the code. This then re-numbers the boxId_e modes.
In the Modes.md doc
https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md
the mode listed are 0,1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21. 
The modes.md doc then says:
13 	BEEPERON 	Enable beeping - useful for locating a crashed aircraft
which is false. This is due to what was #4 being commented out so BEEPERON is now #12.

The cli aux command to assign an AUX channel to the beeper according to the doc is:
aux 4 13 1 1700 2100 (aux channel #1 when between 1700 and 2100 turns on the beeper).
This does not work since the boxId_e enum has #4 commented out.
The cli command that actually works properly is:
aux 4 12 1 1700 2100
This is what I mean by bad programming. The change by commenting out #4 broke the code that uses the boxid_e enum and the document.
A better way of removing the #4 mode would have been to replace it with a 'dummy' name. This way any code or doc would not require a change.

tracernz, 2015-02-22:
To clarify the issue:
Using the MSP commands works perfectly and the box IDs listed in Modes.md work correctly. NO problems with the MSP implementation. See https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L304-L329 where the MSP Box IDs are mapped as per Modes.md.

Using the CLI `aux` command, all IDs >3 are offset by -1 compared to Modes.md, this is caused by BOXVARIO being commented out in https://github.com/cleanflight/cleanflight/blob/master/src/main/io/rc_controls.h#L22-L47.

I feel the best way to fix this is to keep the enum in rc_controls.h aligned with the MSP permanent IDs, as per my PR #510. The downside of this is that it will break existing configs.

hydra, 2015-02-22:
How about just adding another column in the documentation for the cli values or updating the CLI commands to use the permenant IDs?

tracernz, 2015-02-22:
Updating the CLI commands to use the permenant ID would also be a good solution, and probably the best one actually. I think leaving as-is and adding another column with another set of numbers will just add to confusion in the future.

hydra, 2015-03-09:
Ok, I'm not happy with any of the proposed solutions, the BOX* should be decoupled from multiwii/msp - cli should not have a dependency on MSP.

Code should NOT be written to parse any CLI commands, it is not meant for machines, it is meant for users.

Since there are no, and should be no, CLI clients other than users it doesn't make sense to force all users to update the configurations.  Documentation is for users.  Documenting the CLI is fine.

I will update the documentation to reflect how the cli works to avoid confusion and close this and the PR.

We can re-open this if new information or a better solution comes up.

closed 2015-03-09 by hydra