cleanflight/cleanflight #521: I2C support on CC3D broken

digitalentity, 2015-02-18:
I have a BMP085 breakout board connected to CC3D FlexiPort, however cleanflight does not recognize it. UART3 is unused in ports configuration tab. I've tried with cleanflight 1.7.1 & 1.7.2 with no luck.
This breakout board works fine and reports valid data with test code on mbed.

hydra, 2015-02-18:
I2C support is not broken, my OLED display works just fine.  Check the I2C signals are clean using a scope and try using different pull-up resistor values.

If it's really broken please reopen.

digitalentity, 2015-02-18:
Thanks for updating the title.

I don't have a scope, so I can not verify the I2C signals, but I'll try to play with pull-up resistors and try to confirm the issue.

digitalentity, 2015-02-19:
@hydra, I've connected a logic probe to I2C bus and recorded the data exchange while board boots. It looks like cleanflight is able to communicate with I2C device 0x77 (BMP085), reads chip ID (reg 0xD0), reads version (0xD1), reads calibration (0xAA). 

Looks like detectBaro() should return true. I can not verify it, my CC3D board does not have a SWD connector and I am not very good at soldering SMDs.

I looked at the code and it occured to me that sensorsAutodetect function is missing sensorsSet(SENSOR_BARO) at line 390 (right after bmp085Detect). Since CC3D does not have a barometer onboard, SENSOR_BARO bit is not set in SENSORS_SET.
Current code never sets SENSOR_BARO, so external barometer never gets detected and used.

I believe same thing is true for external magnetometer.

digitalentity, 2015-02-19:
Please see screenshot of reading BMP085 chip id:
[image]

digitalentity, 2015-02-19:
I also created a proposed patch for 'src/main/sensors/initialisation.c' file:

@@ -396,12 +396,14 @@
 #ifdef USE_BARO_MS5611
     if (ms5611Detect(&baro)) {
+        sensorsSet(SENSOR_BARO);
         return;
     }
 #endif
 #ifdef USE_BARO_BMP085
     if (bmp085Detect(bmp085Config, &baro)) {
+        sensorsSet(SENSOR_BARO);
         return;
     }
 #endif

digitalentity, 2015-02-19:
I've forked you repo and will try to fix this myself. If my patch will work, I'll create a pull request.

Can you please reopen the issue as I am unable to do it.

digitalentity, 2015-02-19:
I was able to fix the issue - now external baro (BMP085, BMP180) & magnerometer (HMC5983) are correctly detected and used if connected:
[image]

I've created a pull request with the fix.

hydra, 2015-02-19:
Yes I saw t. Great, thanks for looking into it. I will get that merged asap.

hydra, 2015-02-19:
#525 is merged and i did further cleanup work.  Please checkout the latest master and re-test.  Re-open this if there's still an issue.

closed 2015-02-19 by hydra

Tomal2v, 2017-01-16:
how to solve this BMP180 detection problem on cc3d