cleanflight/cleanflight #523: ADC Support on Sparky for Voltage/Current Monitoring

maximan, 2015-02-18:
The Sparky port has come on leaps and bounds, but the inability to monitor voltage and current onboard is a major limitation (no voltage on OSD, battery warning via buzzer/LED, etc). Implementing ADC support would effectively bring the Sparky port on a par with Naze32, CC3d etc.when used in conjunction with a voltage (and current) sensor.

hydra, 2015-02-18:
yup, the ADC code for the F3 is written just not enabled for the sparky yet, shouldn't take someone long to add it and test it.

for an example, see here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/target/SPRACINGF3/target.h#L87-L107

maximan, 2015-02-18:
I am more than happy to test this code and have a voltage and current sensor ready to go - if someone is able to add it. :-)

MJ666, 2015-03-06:
Just working an VBat support for the AlienWii F3 which is quite similar to Sparky. Two pins which can be configured as ADC input and using the same ADC can be configured for voltage and current monitoring.
Here I would recommend PWM8 (current) and PWM9 (voltage). This will allow at least voltage monitoring even with an Octocopter. PWM10 is also capable as an ADC input but is part of an different ADC and not easy to configure together with PWM9. I can make an test build for this. @maximan not sure how do we get this easily to you?

maximan, 2015-03-06:
@MJ666 I have already gone for GPIO pins 1 and 4, which correspond to PWM 9 and 10 on the Sparky board and have a build to test (but won't be able to test until next week). I think the trickier proposition is going to be adding support for BUZZER and SONAR - although I think the solve for SONAR will be the same PWM5 and PWM6 solution used for Naze. DISPLAY is quite a bit more work also, but arguably less useful than BUZZER and SONAR.

MJ666, 2015-03-06:
@maximan PWM9 and PWM10 will not play together for ADC. You should use the pins as described above. This would require a major code rewrite for the ADC support @hydra may comment this. The buzzer can be configured to any available IO pin. Unfortunately they are not easily accessible (only via soldering to the STM directly. May be using PWM10 will be an option but I need to take look at the code if it can be freed up from the PWM output also like for ADC. The BUZZER can only be configure during compile time. One option would be to remove the last PWM output completely. you also need to add the drivers (resistor and transistor) for the BUZZER externally. My understanding was you wanted to go for telemetry anyhow?

maximan, 2015-03-06:
I went for PWM9 and 10 as this is what TauLabs are using, so I assumed these would be okay? The schematic shows the inline resistors on these pins: https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky/hw/Sparky_v1.0.pdf

Telemetry is already working, I believe. I don't mind a bit of soldering to add a buzzer - although this issue is related that would offer an alternative: https://github.com/cleanflight/cleanflight/issues/199

MJ666, 2015-03-06:
@maximan I agree from the hardware it will work. Only the actual ADC driver implementation of CF will not support this as it is an need some major changes.to support ADC cannels from multiple ADC's. PWM10 is on ADC1 and PWM8/PWM9 are at ADC2.

JohnieBraaf, 2015-04-12:
This issue may be closed, see https://github.com/cleanflight/cleanflight/pull/635

hydra, 2015-05-15:
Done

closed 2015-05-15 by hydra