cleanflight/cleanflight #529: Add support for ADNS3080 Optical Flow Sensor

matbogdan79, 2015-02-19:
Hello,
Please add support for ADNS3080 Optical Flow Sensor.
I am 3d printing my DJI Inspire drone and i like cleanflight repo.
ADNS3080 Optical Flow Sensor will help hold the drone in place even with low GPS signa or indoor.
This addon will make CleanFlight beat the DJI software in features.

Thank you and hope i will hear about implementation of the ADNS3080 Optical Flow Sensor very soon.

digitalentity, 2015-02-19:
I second this. It would be a useful feature for novice pilots doing indoor or low-height flights.

tracernz, 2015-02-20:
http://nz.mouser.com/ProductDetail/Avago-Technologies/ADNS-3080/?qs=RuhU64sK2%252bsop%2F60ZpFX6g%3D%3D
Looks like it's an obsolete part? I can't find it available anywhere..

matbogdan79, 2015-02-20:
try this: http://www.aliexpress.com/item/ADNS3080-Optical-Flow-Sensor-Horizontal-Position-for-APM2-5-Flight-Control-Board/1404445315.html
I've ordered mine one week ago so next week should be in my hands and first i will connect it with my arduino.

dkisselev, 2015-02-21:
Looks like 3DR also sells them, so they're at least somewhat used in the community: http://store.3drobotics.com/products/optical-flow-sensor

matbogdan79, 2015-02-21:
DJI use them in their DJI Inspire transformer.
As I said before I am 3d printing one. I wold love to have this optical sensor added so practically i can fully clone an transformer.
[image]
[image]

digitalentity, 2015-02-22:
@matbogdan79, DJI might use something different from ADNS3080, something custom-built (like http://store.3drobotics.com/products/px4flow). However ADNS3080 is widely available from China (aliexpress, banggood etc), so support for it is worth implementing.

tracernz, 2015-02-22:
All of the links provided above for the 3DR ADNS3080 based board are out of stock. There may be some small stockpiles around somewhere but it seems they are running out. The sensor IC itself was discontinued in 2012 when Avago shutdown that division of their business.

There are a few relevant research papers around which I have found interesting (e.g. http://www.araa.asn.au/acra/acra2010/papers/pap139s1-file1.pdf). Does anybody know what kind of range the 3DR board is capable of working effectively at? The sensor developed in the paper linked above was only useful for less than 0.4 m.

matbogdan79, 2015-02-22:
The sensor is avaible:
http://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20150222023845&SearchText=ADNS3080+Optical+Flow+Sensor

This sensor it has been discused already on multiwii forum: 
http://www.multiwii.com/forum/viewtopic.php?f=7&t=1413&start=20#p11421

This sensor should work with good accuracy up to 4-5 meters.
The closer you are to the ground the better accuracy.
After 4-5 meters you can switch to GPS-baro because also the sonar will also not help you.
So practically you can do:
- sonar + optical flow sensor for possition hold under 4-5 meters
- GPS+Baro after 4-5 meters.

Of course the optical flow sensor will not help you return to home :) but will hold your drone in place better than the GPS when you are under 4-5 meters or indoor.

matbogdan79, 2015-02-22:
Here is some explanation of what and how this sensor should work:
https://www.youtube.com/watch?v=b0WtqaiM20E#t=3893
Sonar+optical flow sensor.

dkisselev, 2015-02-23:
Just found this as well: http://www.hobbyking.com/hobbyking/store/__66308__HK_Pilot32_Optical_Flow_Kit_With_Sonar.html adds sonar in a nice package.

I'd love to work on this. Don't count on it, but if I find some spare time to contribute code to CF in the next couple of months, this will probably be the project I work on.

matbogdan79, 2015-02-23:
@dkisselev very expensive. 
Alternate you may get them separated:
Sonar: http://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20150222222511&SearchText=Ultrasonic+Module+HC-SR04

Optical Flow: http://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20150222023845&SearchText=ADNS3080+Optical+Flow+Sensor

Why pay 99 USD+shipping when you can pay under 20USD for both plus that you may chose where to place them.

dkisselev, 2015-02-23:
Pixflow looks like a much higher quality sensor (and has processing room for machine vision if that's what you're into), but yeah, can't really beat the price of the ADNS3080). Just saying that CF should support both because some people might have px4flow's from leftover projects or want more precise measurement.

marksev1, 2015-02-25:
Adns3080 is also very readily available on eBay. If a IC is being discontinued, it not a problem for the Chinese, since they just clone it...an example for that is max7456 for minimosd :)

Hardwiring, 2015-02-28:
Optical flow sensor support would be awesome, anyone seen the parrot rolling spider? Horrible Bluetooth thing, but position hold to kill for. Is this to much to ask? I imagine there would be a lot of processing involved. These modules seen to take a low resolution image and output something far more complex than the signal from the sonar unit.....

matbogdan79, 2015-02-28:
Optical flow sensor , get an low res image with high FPS and with some image processing ( matrix processing ) can give you left/right forward/backward movement with high accuracy on low altitude and with good light conditions. The sonar is other axe of movent up/down.
Processing the image should be possible on 32bit proceesor because it is a low resolution.

Here is some example of the data is output by the optical flow sensor: 
https://www.youtube.com/watch?v=yglonHYYoSg
https://www.youtube.com/watch?v=vXPdUOgVKiA

here si some code to play with and get inspired by:
https://github.com/Trexter/Autonomous-Arduino-Quadcopter

Considering it works on arduino, should not be any problem using it on powerful naze32 and derivatives..

dkisselev, 2015-03-02:
Optical flow sensors come with their own processing chips built in, the ADNS3080 that we're talking about basically makes the motion vector available on an SPI bus that updates a certain speed (can't find the exact sped right now, but it's fast enough. PX4Flow is similar but uses i2c and outputs the data in a slightly different way).

All that needs to be implemented on the Cleanflight side is taking that vector and using it to apply a navigation correction to the flight pattern.

Unfortunately, Cleanflight doesn't really have a solid navigation framework built yet, most of the places that affect flight patterns (barometer, sonar, GPS) each calculate their own correction values and override the RC input values to bluntly push the quad in the right direction.

In Ardupilot, this problem is solved by using a latitude/longitude coordinate system for core navigation. The optical flow sensor bumps the lat/long values based on the last movement vector reading, and the core navigation framework tries to reset those readings to 0 (or the last 'hold position' GPS point if the GPS is available). CF has some similar navigation PID's in the GPS code, but that's still legacy from multiwii and not very readable or expandable.

shodanx2, 2016-04-25:
It's a year later, this sensor is more available than ever and cheaper than ever at 14$ it might even make sense to have two or three of them held at right angles !

At this time, the cheapest of multiple sources on ebay is this one
http://www.ebay.com/itm/CJMCU-110-Optical-Flow-Sensor-for-APM-2-52-APM2-6-Flight-Controller-e-/261879391405?hash=item3cf93a64ad:g:NScAAOSw-W5Ux0o8

solid arduino sketch implementation available at 
https://github.com/impressivemachines/Arduino
with youtube demonstration of the output
https://www.youtube.com/watch?v=o4IfkTGKw3c

it requires very little computing power the output is about 10 hertz and consist of 
surface quality (integer) 0 to 63
average pixel brightness integer) 0 to 63
exposure integer 0-9100
motion vector in the form of  (1,-1)  or (-21,10) etc.  two signed integers -63 to 63, give the x and y motion since last frame

Linjieqiang, 2016-05-05:
nice,what about use px4flow?

shodanx2, 2016-05-06:
There is no question that the px4flow is a better sensor however it is in
short supply, a specialty item.

Using the px4flow is much much computationally demanding and the code to do
this will be complex.

Lastly the cost of the px4flow put it outside the budget of most users, at
150 euros it costs more than my entire quad.

By comparison the adns 3080 is 14$usd including shipping and it widely
available by more than 10 distributors.

The output of the sensor is so simple or can be understood by a human in
real time, the important part being an XY vector representing movement,
like.

-5+8 , 5 points to the left, 8 point forward

How much distance those points represent is proportional to altitude.

The control code is then very simple.

Combined with a 5$ sonar module to provide altitude hold, this sensor
provides simple but effective position hold.

johnsilvester, 2016-09-08:
Has anyone made any steps in implementing ADNS3080?

martinbudden, 2016-09-08:
Yes, @digitalentity is working on it here: https://github.com/iNavFlight/inav/issues/216

However this is still in fairly early stages, so I wouldn't count on it for a school project.

closed 2017-02-25 by hydra

meherranjan, 2017-11-29:
Are there any updates on the development side of this. I read through the iNavFlight and it still is far from complete. As mentioned earlier the sensors are abundantly available in the market and they make up a good addition to flight experience. Cleanflight is a great project and I don't see any reason to not include this. Is it kept on hold for a reason? Are there other priorities that are impeding development on this? Or is it project on a future roadmap? I would love to get some insights :)